SanPen/VeraGrid #2: constant current loads in dSbus_dV

jkrstulo, 2017-04-05:
Dear Santiago,
my name is Jakov Krstulovic Opara, and I am one of the contributors to pandapower. 
we have been implementing zip loads to pandapower pf, so I had a look at your code and I came into a doubt when looking at consideration of constant current loads in the dSbus_dV function (specifically [here](https://github.com/SanPen/GridCal/blob/master/UnderDevelopment/GridCal/grid/IwamotoNR.py#L77)):
so for sparse Ybus version there is: `Ibus = Ybus * V - I`, while for non-sparse ([here](https://github.com/SanPen/GridCal/blob/master/UnderDevelopment/GridCal/grid/IwamotoNR.py#L83)) there is: `Ibus = Ybus * asmatrix(V).T + I`
according to what it seems to me and according to sing of I for a load, I would say it should be + ?

Thank you and sorry for bothering you with this...also sorry if I misunderstood something...

also I need to say my great congratulation for GridCal - it is really a great tool which was missing in open-source of power system analysis...it is fascinating you did all by yourself! great work!

SanPen, 2017-04-05:
Hi, thanks for the compliments!

I have just double checked, and the implementation makes sense as it is now.

To clarify the signs:
- The current in a load in the interface is positive for load effect and negative for generation effect. This means that if I create a load of 10kA it should make the voltage drop, and if I enter a load of -10kA it should make the voltage raise.

- Now under the hood, both power (`S`) and current (`I`) are summed into the `Sbus` and `Ibus` arrays with negative sign, hence my 10kA load would be `-10kA / (Vnominal * sqrt(3))` at the `Ibus` vector

- The `Ibus` vector (with presumably negative values for loads in it) is passed to the power flow algorithms.
  Now in the Jacobian, where that formula belongs:  `Ibus = Ybus * V - I `. The Ibus from the Jacobian formula is expected to be positive, that is why I add the currents injection vector with negative sign.

I hope It is clear. Nevertheless, if yu think there's an error, please let me know to fix it. My reasoning is that at the moment, the results make sense for the used signs.

jkrstulo, 2017-04-06:
Hi, thank you for so detailed explanation!
ok, minus sign in `Ibus = Ybus * V - I` is now clear, but still I don't get why in case of non-sparse Ymatrix you have + sign [here](https://github.com/SanPen/GridCal/blob/master/UnderDevelopment/GridCal/grid/IwamotoNR.py#L83) ?

SanPen, 2017-04-06:
A bug probably, I'll fix it now

jkrstulo, 2017-04-06:
ok :) that is actually that made me a confusion and I was no more sure what was correct. At least we found a tiny bug :)
thanks a lot once more...your software is really great, I'll definitely continues using it and may come back to you with some questions.
best regards

closed 2017-04-06 by SanPen