SanPen/VeraGrid #3: Get some warning in Windows 10 ,Python 3.6.1

fxt0706, 2017-05-03:
Hi, Santiago

I'm a student in China and now using GridCal for my learning of Grid, thanks for your great tool!

As the title, here are some errors in my computer system, Windows 10, Python 3.6.1.

When I run the python code like:

```
from GridCal.ExecuteGridCal import run
run()
```

It will report that:

> Warning (from warnings module):
>   File "D:\Program Files\Python36\lib\site-packages\matplotlib\__init__.py", line 1405
>     warnings.warn(_use_error_msg)
> UserWarning: 
> This call to matplotlib.use() has no effect because the backend has already
> been chosen; matplotlib.use() must be called *before* pylab, matplotlib.pyplot,
> or matplotlib.backends is imported for the first time.
> 
> 
> Warning (from warnings module):
>   File "D:\Program Files\Python36\lib\site-packages\matplotlib\cbook.py", line 136
>     warnings.warn(message, mplDeprecation, stacklevel=1)
> MatplotlibDeprecationWarning: The axisbg attribute was deprecated in version 2.0. Use facecolor instead.
> 
> Warning (from warnings module):
>   File "D:\Program Files\Python36\lib\site-packages\GridCal\grid\CalculationEngine.py", line 2099
>     warn('The file does not exist.')
> UserWarning: The file does not exist.

I just updated the GridCal, don't know whether the Mac or Ubuntu will report like that too.

The other problem is when I run with the user interface. After I dragged some buses and set some value, I clicked the Save Button and clicked the Save in the explorer window appeared, it will prompt stop running and had no error report.

I'm not quite sure how this happened, but when I opened your sample xlsx file, it would not seem to have the same mistake.

SanPen, 2017-05-03:
Hi, 

I use Ubuntu and I get the warning too. I am aware of it and will work on the solution sometime soon.
I hope it is not a critical issue in your computer.

About Mac / Ubuntu / Windows I expect all of them to behave well with GridCal.
Under windows I certainly advise you to use python through Anaconda. Other than that it should be ok.

Best regards,
Santiago

SanPen, 2017-05-04:
Solved it. 
Will be fixed in the next update.
Cheers!

fxt0706, 2017-05-10:
Happy to hear it!
Waiting for next update XDD

SanPen, 2017-05-10:
I have been not able to get a crash when creating a grid and saving later. 

(I fixed the warning and I forgot about the crash...)

Would you be so kind of detailing the precise steps that lead you to crashing?

PS: You may update, maybe it is solved already. The version as of today is 1.41 (check  help>About)

fxt0706, 2017-05-11:
Hi, Santiago

Tomorrow I will update it and try to create grid again. If the crash still existed I will record it as a short gif to show how it get a crash. Hope it already solved!

Thanks for your reply!

SanPen, 2017-05-17:
Hi, I have been testing in Windows with the latest Anaconda (python 3.6)
Indeed sometimes I get a clean crash when saving.
When I try to debug I get no error.

This looks like a problem with Pandas under windows. Try updating Pandas.

fxt0706, 2017-05-21:
Hi, Santiago

Now I'm trying to update Gridcal to latest version 1.47 and follow the command guide to install numba and 11vmlite and get the error 

> Collecting llvmlite
> Using cached llvmlite-0.18.0.tar.gz
> Building wheels for collected packages: llvmlite
> Running setup.py bdist_wheel for llvmlite ... error
> Failed building wheel for llvmlite
> Running setup.py clean for llvmlite
> Failed to build llvmlite
> Installing collected packages: llvmlite
> Running setup.py install for llvmlite ... error

I have to say the Windows is not a good system to do something like that, but tomorrow I will try to fix llvmlite first (maybe just Google it could fix) 

Thanks for your reply and work!

SanPen, 2017-05-21:
install it using anaconda.

something like: `conda install numba`

fxt0706, 2017-05-22:
Hi, Santiago

Thanks for your suggestions and I installed all the support library successfully.  I download the latest version of Pandas from the official site but found it was already lastest.

But this time when I try to save the xlsx file and failed, I got a new report in this version of Gridcal that "ModuleNotFoundError:  No module named 'openpyxl' "

when I installed the openyxl and try to save file again, the problem solved.

When the error occurred the cmd would close automatically in a very short time and all I could do is got a screenshot:

[image]

Anyway, openyxl should be the problem.

After this, I installed a Ubuntu system, I have to say Windows always makes me crazy XD

SanPen, 2017-05-22:
Did your problem solved after installing openyxl?
I thought that package did come with Anaconda...

fxt0706, 2017-05-23:
Yes, The problem solved with openyxl.

I'm sorry that at first I simply use [this site](http://www.lfd.uci.edu/~gohlke/pythonlibs/) to install packages and didn't use the Anaconda(because I didn't see the youtube video guide).

I didn't realize the power of Anaconda until yesterday T-T

Thanks for your patience and help!

SanPen, 2017-05-23:
I'm glad it worked.

closed 2017-05-23 by SanPen