SanPen/VeraGrid #4: Problem installing with Python 3.5

andreqts, 2017-05-04:
After having problems trying to install with Python 3.6, I've created an environment using Python 3.5..
After entering "pip install GridCal", Python crashed (a GPF indeed), and then came this error:

(pyconda35) C:\Anaconda\envs>pip install GridCal
Collecting GridCal
  Downloading GridCal-1.39.tar.gz (297kB)
    100% |################################| 307kB 468kB/s
    Complete output from command python setup.py egg_info:

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 3221225477 in C:\Users\Quites\AppData\Local\Temp\pip-build-_d7i4w0f\GridCal\

andreqts, 2017-05-04:
I am using Anaconda 4.3.1, 64 bits, on Windows 10.

andreqts, 2017-05-04:
Now I've created an environment from scratch using "conda -n vGridCal python=3.5 numpy pandas matplotlib"

it also did not work :(

Follows the error messages from my console:

(vGridCal) C:\Anaconda\envs>pip install GridCal
Collecting GridCal
  Using cached GridCal-1.39.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\Quites\AppData\Local\Temp\pip-build-s8jq1rt1\GridCal\setup.py", line 5, in <module>
        from GridCal.grid.CalculationEngine import __GridCal_VERSION__
      File "C:\Users\Quites\AppData\Local\Temp\pip-build-s8jq1rt1\GridCal\GridCal\grid\CalculationEngine.py", line 28, in <module>
        import networkx as nx
    ImportError: No module named 'networkx'

Thanks in advance for your help and congratulations for your great project!

andreqts, 2017-05-04:
I´ve tried to install networkx before, looks what happens:

(vGridCal) C:\Anaconda\envs>pip install networkx
Collecting networkx
  Using cached networkx-1.11-py2.py3-none-any.whl
Collecting decorator>=3.4.0 (from networkx)
  Using cached decorator-4.0.11-py2.py3-none-any.whl
Installing collected packages: decorator, networkx
Successfully installed decorator-4.0.11 networkx-1.11

(vGridCal) C:\Anaconda\envs>pip install GridCal
Collecting GridCal
  Using cached GridCal-1.39.tar.gz
No files/directories in C:\Users\Quites\AppData\Local\Temp\pip-build-_ob408ao\GridCal\pip-egg-info (from PKG-INFO)

(vGridCal) C:\Anaconda\envs>pip install GridCal
Collecting GridCal
  Using cached GridCal-1.39.tar.gz
    Complete output from command python setup.py egg_info:

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 3221225477 in C:\Users\Quites\AppData\Local\Temp\pip-build-kyn25egt\GridCal\

SanPen, 2017-05-04:
Hi Andrew, 

I have GridCal working reports from people with W10 and Python 3.6 from Anaconda.
I would advise you to stay with the default python. 

That said, I believe you might have  a permissions issue.
Would you mind running the console as administrator in your system?

On top of that, I advise my co-workers and friends to always install Anaconda "for all users" and do it in `C:\Anaconda3`, not in the program files folder.

I have tested GridCal with blank , windows systems and just installing Anaconda and it works on the first try.

So please, retry with admin permissions, and if not, uninstall All possible Pythons you might have and install the latest Anaconda Python on C:\Anaconda3

andreqts, 2017-05-04:
Thank you for your return.

I'd already tried running as admin and got the same result.

Ive just seen some cases in stack overflow from people having installed different versions of Anaconda, what I think I did when I began using it, as a newbe. Maybe that has something to do with my problem.

I will try to reinstall it and see if it solves my problem.

Thanks!

SanPen, 2017-05-04:
Good luck, and please write back if you solve the issue.

SanPen, 2017-05-06:
I recorded this: [https://youtu.be/yGxMq2JB1Zo](https://youtu.be/yGxMq2JB1Zo)
It is all straight forward. So I am convinces there is something with the permissions in your computer.

andreqts, 2017-05-10:
Ok Santiago, I've successfully installed it both on Windows 10 and Linux Mint, with a clean Anaconda3 installation. In Anaconda2 I've got problems, even using Python 3.

The problem was indeed with my installation.

Thanks for your help.

The only things I needed to do differently seems to be because I have created a conda environment to GridCal. So if I run "pip install GridCal" directly it fails. I had to firstly install numpy, matplotlib (these first two I already created the environment containing it), and then scipy, pandas and networkx, and only after installing it I was able to effectively install GridCal. But it is now working perfectly.

So I would leave as a suggestion to make examples installing in a conda environment, since it is a good practice.

Thank you very much for sharing your work.

closed 2017-05-10 by andreqts

SanPen, 2017-05-10:
noted ;)