SanPen/VeraGrid #5: CIM support

stv0g, 2017-08-07:
Dear San,

are there currently any plans to add support for CIM?

Thank you

SanPen, 2017-08-07:
Hi, 

You may know that CIM is not strictly a standard, hence everyone implements it as they prefer.

Besides CIM "standard" contains classes for absolutely every tiny object in the grid and its implementation is mayhem. I know this because we support it at work.

So in short, I have no CIM import planned because CIM is not a real standard, is a good will template.

For transmission I consider instead to support PSS\E, which is kind of fixed and everyone uses it.

So, if you have the strength and heart to try to implement CIM for GridCal, then I can help you ;)

stv0g, 2017-08-07:
Hi @SanPen,

Thank you for the response :)

I know. This magnitude of classes within the ENTSO-E CIM profiles is mind boggling.
To cope with the number of classes, we use automatic code generation tools to transform the UML model into C++ / Python code.

But still, I think CIM (including all its profiles) is a family of standards which precisely specifiy how a CIM file should look like.

We are currently trying to do load flow analysis on CIM models.

SanPen, 2017-08-07:
Hi, 

Indeed we have this Enterprise Architect project to generate C++ to implement CIM, but still the amount of classes inheritance and dynamic casting is just too much.

To deal with it you need to have some sort of mapping. If you'd like to use GridCal as engine, a sufficient hello world example is [this](https://github.com/SanPen/GridCal/blob/master/UnderDevelopment/GridCal/tests/DEMO_5_node.py).

Once the `MultiCircuit` object is populated, you can access to all there is inside GridCal.

SanPen, 2018-08-04:
CIM Export included. I'll try to deal with import some time soon.

aselviar, 2019-03-14:
Hi

Any update regarding CIM import?

SanPen, 2019-03-14:
Done

But this would be my interpretation of CIM, which as you may know is way less standard than say PSS/e.

miek770, 2019-03-17:
In case someone's interested, I asked OTI to support CIM imports in ETAP (they'll think about it...) and to fix their broken PSS/E import. I also asked EasyPower to support CIM. I don't have huge expectations, but I'll test it (CIM) if either makes it available.

SanPen, 2019-03-17:
I can imagine that CIM import could be a nightmare for a software company. Unlike a normal standard that sets the "recommended" way of doing something, in CIM there is a trillion ways to do the same thing. And the objects structure is incredibly convoluted. So if ETAP implements it, they'll get many ticked claiming that the CIM import doesn't work, just because the user is trying to import a CIM file from another company who interpreted CIM they way they though it was best.

Furthermore, from version to version they break the retrocompatibility. From v14 to v16 the transformers were completely redesigned to be specified by winding (which kind of make sense)

Lately in the IT world, the idea of CIM is taking traction because the management layers in utilities are fond of the idea of an IT design that can take anything. But again every utility will implement CIM in its own way. It is like the XKCD comic strip on standards (https://xkcd.com/927/).

So, when I implemented CIM in GridCal, it was my interpretation of it. I guess the key is to set an implementation guideline. For GridCal I used the way that Digsilent implemented it by reverse engineering their CIM export files.

miek770, 2019-03-18:
Hehe XKCD is gold.

I didn't realize CIM was so loose as a standard. BTW, I asked some questions about an open, common data format on the Openmod forum, see here: [https://forum.openmod-initiative.org/t/standard-open-common-data-format-cdf-for-power-systems/1169](https://forum.openmod-initiative.org/t/standard-open-common-data-format-cdf-for-power-systems/1169). Based on your responses @SanPen, I'd say a truly open CDF for easy power system model exchange between softwares remains a simple (or not so simple) wish for now.

PSS/E's format's probably the next best thing.

SanPen, 2019-03-21:
@stv0g , @aselviar  Any feedback? Otherwise I'll close this issue.

aselviar, 2019-03-21:
Hi SanPen

Thanks a lot for your comments and for re-opening the issue.

I have a CIM export from a small IT company that have their own interpretation of CIM that I wanted to import to GridCal, but I unfortunately cannot share any more information at the moment, since the model is one of a real grid.

I did not know that there were so many different interpretations of CIM.

I am sorry that I cannot help any further at the moment and would come back to you if I have anything new.

SanPen, 2019-03-21:
Then I suspect that you are already aware of the problem of CIM not being a real standard.

I suggest that you modify my implementation to read you files.  If it does not break, the GridCal CIM importer will create a `CIMCircuit` object with all the objects inside. There you may parse those as you consider it suits the format. The file is [this](https://github.com/SanPen/GridCal/blob/master/src/GridCal/Engine/Importers/cim_parser.py)

Good luck!

closed 2019-03-21 by SanPen