SanPen/VeraGrid #9: RuntimeError: cannot cache function 'create_J': no locator available for file

fxt0706, 2017-11-02:
Hi Santiago,

I tried to use `pyinstaller ` with `ExecuteGridCal.py` to output the .exe file for Windows, and it built successfully.

But when I tried to run the ExecuteGridCal.exe, it shows the error below:

> E:\PythonProject\GridCal\UnderDevelopment\GridCal\dist\ExecuteGridCal>ExecuteGridCal.exe
Numba was detected, enjoy :D
Traceback (most recent call last):
  File "GridCal\ExecuteGridCal.py", line 21, in <module>
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "D:\Anaconda3\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 631, in exec_module
    exec(bytecode, module.__dict__)
  File "GridCal\gui\main\GridCalMain.py", line 16, in <module>
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "D:\Anaconda3\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 631, in exec_module
    exec(bytecode, module.__dict__)
  File "GridCal\grid\CalculationEngine.py", line 18, in <module>
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "D:\Anaconda3\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 631, in exec_module
    exec(bytecode, module.__dict__)
  File "GridCal\grid\JacobianBased.py", line 176, in <module>
  File "site-packages\numba\decorators.py", line 192, in wrapper
  File "site-packages\numba\dispatcher.py", line 490, in enable_caching
  File "site-packages\numba\caching.py", line 594, in __init__
  File "site-packages\numba\caching.py", line 330, in __init__
RuntimeError: cannot cache function 'create_J': no locator available for file 'GridCal\\grid\\JacobianBased.py'
[3512] Failed to execute script ExecuteGridCal

It seems some errors with `numba` and I have no idea what happened with it. But it wasn't a serious problem and hope you could give suggestion when you are free.

Thank you for your jobs. GridCal is really a nice software!

SanPen, 2017-11-02:
Are you using python3?

fxt0706, 2017-11-02:
Yes, I am using Python 3.6

I may update `numba` tomorrow, if error fix, I will let you know. Thanks for your suggestion.

SanPen, 2017-11-02:
Please try updating GridCal itself.

`pip install GridCal -U`

fxt0706, 2017-11-03:
Hi SanPen,

I updated `numba` and git the latest `GridCal`, but the error was still existed.

my Python is

> Python 3.6.3 |Anaconda custom (64-bit)| (default, Oct 15 2017, 03:27:45) [MSC v.1900 64 bit (AMD64)] on win32

my OS is 

> Windows10 64 bit Home Edition

SanPen, 2017-11-04:
After reviewing the question in detail, I can conclude:

- There is no issue with GridCal since it works fine.

- GridCal cannot be "frozen" into an executable, or at least I cannot since it is super hard to get all the C and Fortran dependencies right in an executable. I remember trying when I started the project and it was not possible for me.

- `create_J` is only declared when `numba` is on the system. Probably the executable demands it to be present regardless of it being used.

- If your aim is to distribute GridCal in a Stand-Alone fashion, I recommend you to get a standalone miniconda distribution into a folder, install gridcal into that Miniconda setup and shil the whole folder with a .bat to run the GridCal launching command.

closed 2017-11-04 by SanPen