SanPen/VeraGrid #12: Unable to install GridCal

bmitrauncc, 2017-12-08:
Hi Sean,

I am trying to install GridCal in Win 10, but I keep facing the error, see screen shot, I have tried all the remedies with regards to it on the internet but nothing seems to work. Need your help...

[image]

SanPen, 2017-12-08:
Hi, unfortunately this is due to a bug in pip (I guess) because it does not install all the required packages upon GridCal setup as it should.

Observe the sentence 

> ModuleNotFoundError: No module names 'pyDOE'

This tells you that pyDOE is missing and you need to install it.

You fix this with `pip install pysot`

Do the same with the rest of the modules that it tell you that are missing.

I'll try to fix this but I have included in the setup file all that the documentation recommends, so I guess this is a bug of pip...

bmitrauncc, 2017-12-08:
I installed all the missing modules, not sure how to handle this one.

[image]

bmitrauncc, 2017-12-08:
Worked after downloading the file from the network...

Thanks for your help

closed 2017-12-08 by bmitrauncc