SanPen/VeraGrid #15: Battery profile

mcanigueral, 2018-05-09:
Hi! I need to simulate a grid with a battery charging and discharging over the day.  
Is there any grid example with a battery and its profile?
How should we specify a charging/discharging profile on a battery component?

Thank you!

SanPen, 2018-05-09:
Hi,

Do you know the battery profile already?
    If that is the case just set the profile in the time events tab and run
the time series (there has to be time series values for load and generation
to make sense) Make sure to check the "dispatch storage" in the settings of
the time series.

Do you want to dispatch the battery?
    run an DC or AC OPF time series and the generators will dispatch
"optimally". Then go to actions and copy those results to the time series
with "Set OPF results as Time events", and run a time series. Again check
the dispatch storage option.

To visualize the profiles, chick on the device and select "plot profiles"

From the provided grids you may use IEEE30. That one has some random
profiles.

PS: I still have to work out on getting storage results...

I hope this helps,
Santiago

mcanigueral, 2018-05-09:
Thanks for the quick response!

Well, I have a grid with different resistive load profiles, and different PV static generator profiles defined. The power of all PV generators are high so during sunny hours appears a surplus of energy in the grid. I want to see the battery capacity necessary to absorbe all this energy surplus and if it could supply the loads during the night.

So, in the profile how can I specify if the battery power in an hour is from charging of discharging?

For example, the first step I want to do: if the grid main branch has the power flux in blue (on sunny hours the flux is reversed due to PV generation) I want to set a battery charging profile before this main branch like the one in grey.

[image]

But I set it on GridCal (with "dispatch storage" checked), and it goes on like this:

[image]

The main branch profile has a lot of reactive power and I don't know where this come from. Additionally I have set battery Qmin and Qmax at zero. Moreover it seems that the battery has not charged.

Is the problem that I want to simulate something that is not implemented in GridCal? Or am I doing something wrong with the data?

And the last thing... I have found that in Grid Cal battery data the energy "E" component appears in kV instead of MWh as it's described in manual:
[image]

Sorry for all this text, thank you very much!
Marc C.

SanPen, 2018-05-09:
Hi Marc, 

First, please update GridCal and see if the units are fixed. In the latest version it is fine (v. 2.25)

If I understand the case correctly;
The PV generator should have a profile of positive values (blue in your first chart).
The battery should have negative values (negative: charge, positive: discharge)

However if you want the battery dispatched automatically with the condition that you described, that is not implemented yet off the shelf.
Of course you could use GridCal as an engine for a custom procedure.

Setting  Qmin and Qmax of a generator to zero will produce weird effects. If you want to forget about the reactive limits, set Qmax to a high value,say 9999 and Qmin to a negative value like -9999. Alternatively you can un check the "control reactive power" option in the power flow settings.

Did it help?

mcanigueral, 2018-05-09:
Sorry, it's not about the battery issue, but I can't upgrade to 2.24 because the PySide package it's not compatible with python 3.6. Do you know any solution about this?
If it's only for the "E" units I will continue working on version 1.98.

About my grid, I have done the next changes:
- Qmin = -9999
- Qmax = 9999
- "Control reactive power" unchecked too
- Battery profile with negative values: -0.01MW from 8:00 to 14:00

Simulation: same results, high reactive power profile in the branch before the battery, and the real part has the same values than without the battery.

However, I think that something I have described in the "battery" sheet is wrong, because with it I can't save the project and appears this error message:
[image]
Before adding the battery I could save the project and simulations were OK.

So, is there any impossible value or maybe the "kV" problem is related with?
[image]

Thank you again!

SanPen, 2018-05-09:
Mmmm the batteries did not really work until recent versions, so I really encourage you to upgrade.
Plenty of what I mentioned is not there in v1.98

Officially GridCal uses PyQt, so if you need to have PySide on your python setup, I'd recommend you to install the [standalone version of gridcal](https://mega.nz/#!Dl5m0b7T!brD9O0ejs2Ih6rOLVaCjQm3-JZttH-ksrp5xXrrX3BU) and upgrade from there.

Regardless, of this, what is the error message you're getting when trying to upgrade?

The upgrade command should be `pip install GridCal --upgrade`

mcanigueral, 2018-05-09:
Great, now I have 2.24 version installed. The error was that I did not have PyQt package.

However I'm still having the same problems with battery data:
1) I can't save the file (error dialog of the comment I posted before)
2) When I want to Display the batteru profile on Time events tab, Python stops working and Windows has to close it.

I suppose that the simulation error it's because of this... I don't know. Any idea?
Thanks!

SanPen, 2018-05-10:
At this point we can do two things:

a) Send me the grid and I try to find the problem.
b) please explain a minimal case so I can reproduce and fix the issue.

Thanks

closed 2018-05-20 by SanPen