SanPen/VeraGrid #17: Simple CLI examples

miek770, 2018-09-21:
Hi,

I would like to suggest adding basic CLI examples for working with gridcal without the GUI, for example in a Python script. I tried to find some in the manual and the wiki but didn't.

Also, a mailing list it a forum would be nice, and a probably better place to ask questions than a GitHub issue.

Thanks.

SanPen, 2018-09-21:
Hi Michael, 

Indeed some didactic examples would be nice.

The place I have with CLI examples is the file: [Api_test.py](https://github.com/SanPen/GridCal/blob/master/UnderDevelopment/tests/API_test.py)

At the moment I am focused at re-writing the core to enable a much faster circuit compilation to enable reliability studies, so my time to do this is limited.

So, more hands are very welcome.

SanPen, 2018-10-04:
Hi, 

I think that a good place for the examples would be to include them in jupyter notebooks or just plain .py files in the tutorials folder.

The manual is outdated and would be nice to have the examples there as well.
LaTex ca be installed with [MikTex ](https://miktex.org/download) in windows
and then you need an editor like [TexStudio](https://www.texstudio.org)
To extend the document should be easy, since the structure is there already.

Indeed the project page could be better...TBD

closed 2019-01-29 by SanPen