SanPen/VeraGrid #21: Transformer types

miek770, 2018-09-27:
I'm wondering what steps must be taken to fully consider the transformer type when running a non-linear load flow.

For example, in my network I have several transformers, all linked to one of two transformer types. With these lines, following the power flow calculation, I can confirm that the transformers were effectively linked to their respective type:

```
logging.debug("Transformers:")
    for branch in grid.branches:
        if branch.branch_type == BranchType.Transformer:
            logging.debug(f" - {branch}: {branch.type_obj.LV_nominal_voltage}/{branch.type_obj.HV_nominal_voltage}, {branch.type_obj.Nominal_power} MVA")
```

However, looking at the calculation results, it is clear that the HV and LV nominal voltages aren't really considered. It looks like the transformer's ratio is always considered equal to the ratio between its HV and LV buses voltages, and maybe its tap module (haven't tested this yet).

Is this feature simply not fully implemented yet? Or do I need to take some other step to ensure that the transformer type's attributes are taken into account, once they are linked to each transformer?

In the calculation engine's MultiCircuit.compile(), should we change the circuit.tap_mod[i] to consider the transformer's nominal ratio instead of simply considering the tap_module? For example:

tap_mod = tap_module * (xfo_hv / xfo_lv) / (bus_hv / bus_lv)

For example, a 245-34.5 kV transformer at its +2.5% primary tap would end up with a tap_mod of 1.046.

Thank you.

SanPen, 2018-09-28:
Hi!

There might be a piece of information that you should know;

When calculating in per-unit (normalization of the grid) a transformer is modeled in exactly the same way as a line. Therefore GridCal does have a generic Branch model to handle all the connection types (Lines, Switches, transformers, reactances, etc...)

The branch_type property is used in higher level functionalities like assigning a branch template (a tower or a transformer type)

So, in the computations, a transformer (or any other branch) terminal voltages are irrelevant.

I hope this answers the question

miek770, 2018-09-28:
Hum I'm sorry but that's definitely wrong, the transformer's ratio absolutely has to be considered, and it's often different from the buses nominal voltage ratio between the primary and secondary. The transformer's ratio should be treated in a way similar to the tap ratio.

If you feed a 25kV bus from a 69-34.5 kV transformer tied to a 69kV network, you'll get 34.5 kV on the LV Side (at no load), not 25kV. If you raise the tap on the primary side by 5%, it'll lower the secondary side by the same percentage.

I can propose a merge request if you want, once we agree on the method.

SanPen, 2018-09-28:
Hi, 

I really advise you to review what I said. A good foundation can be the [ETH course on power systems ](https://pdfs.semanticscholar.org/bdd9/d4b3b0cc625ad81967594398f2a7119d210f.pdf)
(see page 9) Indeed this model is widely adopted in the industry and academia.

Not having the transformers in per unit introduces a numerical discontinuity that would cause the numerical methods to diverge. Hence all voltages, powers and impedances are normalized with the per unit system.

PSS/e, Power Factory and I am sure that the 90% of other power system solvers use this. The ones that do not use the per-nit system use the BFS power flow as formulated [here](https://www.crcpress.com/Distribution-System-Modeling-and-Analysis-Third-Edition/Kersting/p/book/9781439856475), and this method diverges if there are many loops.

Best regards,
Santiago

SanPen, 2018-09-28:
Do you mean the tap changer module?

Changing the tap module in a Branch in GridCal does change the voltage.

The test case IEEE 14 that has some transformers with taps provides the same results as in Power Factory.

  | Voltage module (p.u.) | Voltage angle (deg)
-- | -- | --
Bus 1     HV | 1.060 | 0.000
Bus 2     HV | 1.045 | -4.983
Bus 3     HV | 1.010 | -12.725
Bus 4     HV | 1.018 | -10.313
Bus 5     HV | 1.020 | -8.774
Bus 6     LV | 1.070 | -14.221
Bus 7     ZV | 1.062 | -13.360
Bus 8     TV | 1.090 | -13.360
Bus 9     LV | 1.056 | -14.939
Bus 10    LV | 1.051 | -15.097
Bus 11    LV | 1.057 | -14.791
Bus 12    LV | 1.055 | -15.076
Bus 13    LV | 1.050 | -15.156
Bus 14    LV | 1.036 | -16.034

And the PF results are:
[image]

miek770, 2018-09-28:
Pretty much, in essence it's the same thing. In the end all that matters is that both things (changing tap and changing the transformer's voltages) affect the turn ratio of the transformer. Either should have the same overall impact, which is what I showed in my ETAP example, and what your IEEE example is showing.

So, a 105-10 kV transformer on its nominal tap should yield the same results as a 100-10 kV transformer on its +5.0% HV tap. In both cases, its final turn ratio would be 105/10=1.05*100/10=10.5.

In addition, the resulting voltage in pu will be based on the buses' nominal voltages, not the transformer's nominal voltages. So if the above ideal unloaded transformers are tied to a 100kV bus, their secondary would be at 10*100/105=9.5kV. If the bus' nominal voltage is 10kV, that's be (obviously) 0.95 pu. If it were 9kV, it'd be 1.06 pu.

SanPen, 2018-09-30:
Yes I get what you say.

I guess this would require "virtual tap" devices into the branch model to account for this effect. I take it into account , but as for now I believe the only way to model this is with the tap device.

miek770, 2018-09-30:
Indeed. I might send a pull request to add the feature, probably something similar to what I first suggested above. I'll try to get back to you soon. Let me know if you have new thoughts until then, and thanks for the prompt replies.

miek770, 2018-10-01:
Great thank you, it works. I get the same results using identical setups with either:

1. A 105-10kV transformer (nominal HV tap), tied to a 100kV and 10kV buses;
2. A 100-10kV transformer (+5% HV tap), tied to a 100kV and 10kV buses.

I also like the renaming of "type_obj" to "template", it feels more appropriate.

closed 2018-10-01 by miek770