SanPen/VeraGrid #22: PV control

miek770, 2018-10-01:
Hi,

I'm having trouble getting the PV mode (voltage control) to work. I'm working on a unit test (with pytest) for a simple scenario with voltage control, but it seems like q_min and q_max are set to 0 for non-slack generators, even though this returns the correct limits:

```
for g in grid.get_controlled_generators():
    print(f"Generator {g}: q_min={g.Qmin}, q_max={g.Qmax}")
```
...results in:
```
Generator Utility: q_min=-9999.0, q_max=9999.0
Generator M32: q_min=-2.0, q_max=2.0
```

I have uploaded my test grid as well as its simulation results.

Excel input: [https://www.dropbox.com/s/yyzwpbj2cnu04b8/test_gridcal_pv.xlsx?dl=0](https://www.dropbox.com/s/yyzwpbj2cnu04b8/test_gridcal_pv.xlsx?dl=0)
Excel output (results): [https://www.dropbox.com/s/oqiu2pdazyes5wi/results_test_gridcal_pv.xlsx?dl=0](https://www.dropbox.com/s/oqiu2pdazyes5wi/results_test_gridcal_pv.xlsx?dl=0)

I had a look at the PowerFlowMP.switch_logic() code and it seems to have been rather well developped, so I'm guessing that either:

1. I have made a mistake somewhere in my modeling; or
2. It used to work but some change at some point created a small bug with this feature.

Here's the output from GridCal's logger (from pytest's failed test):

```
---------------------------- Captured stdout call -----------------------------
Padmount transformer updtaed!
Substation transformer updtaed!

Compiling...Trying SolverType.LM
Trying SolverType.IWAMOTO
Control logic
Bus 3  switched to PQ: Q -0.09290827037827094  Qmin: 0.0
Control logic
Bus 3  switched to PQ: Q 2.333504875873943e-10  Qmax: 0.0
Control logic
Bus 3  switched to PQ: Q 2.333504875873943e-10  Qmax: 0.0
Control logic
Bus 3  switched to PQ: Q 2.333504875873943e-10  Qmax: 0.0
Control logic
Bus 3  switched to PQ: Q 2.333504875873943e-10  Qmax: 0.0
Control logic
Bus 3  switched to PQ: Q 2.333504875873943e-10  Qmax: 0.0
Control logic
Bus 3  switched to PQ: Q 2.333504875873943e-10  Qmax: 0.0
Control logic
Bus 3  switched to PQ: Q 2.333504875873943e-10  Qmax: 0.0
Control logic
Bus 3  switched to PQ: Q 2.333504875873943e-10  Qmax: 0.0
Control logic
Bus 3  switched to PQ: Q 2.333504875873943e-10  Qmax: 0.0
```

Thank you, and let me know if you need additional information.

SanPen, 2018-10-01:
Hi, 

Indeed in the new calculation engine there was a bug when copying the Qmax and Qmin limits. That should be fixed now.

I also Added the possibility of visualizing those limits in the GUI in the simulation data tab.

Best regards, 
Santiago

miek770, 2018-10-02:
Thanks, I *think* it works but I've had trouble getting my tests to succeed, and I think I've found another issue (which should be an easy fix).

The Branch object has impedances in pu based on the system's Sb, and in the case of lines, on the line's Vbll. This is what I gather from this description: [https://github.com/SanPen/GridCal/blob/e02f79777fa5251241c966d67c9c0e25321c6498/UnderDevelopment/GridCal/Engine/Devices.py#L791](https://github.com/SanPen/GridCal/blob/e02f79777fa5251241c966d67c9c0e25321c6498/UnderDevelopment/GridCal/Engine/Devices.py#L791)

The TransformerType class accepts a short circuit voltage value in % on the transformer's base, not the system, which makes total sense. When a branch is a transformer, the impedance specified in the Excel input's branch tab are ignored for this device, and the impedances are instead taken from the Branch.apply_type() method, which calls the TransformerType.get_impedances() method. This method, however, doesn't convert the transformer's impedances from its own base to the system's base.

I obviously don't recommend that you change the input method for the transformer type's short circuit voltage, which should remain based on the transformer's bases, not the system.

Or is this conversion happening somewhere further in the code?

Thanks.

miek770, 2018-10-02:
Thank you, [commit 7849f82](https://github.com/SanPen/GridCal/commit/7849f827766f3f383be5d9822286e2fcf21b77c3) seems to have corrected the transformer issue above, and my PV tests are now successful (as well as all others, after some adjustments now that I better understand how you calculate transformer impedances).

closed 2018-10-02 by miek770