SanPen/VeraGrid #23: Transformer with voltage regulator

miek770, 2018-10-04:
Hi,

This is something I wanted to discuss before I submit a pull request. In fact there are 3 reasons why:

1. I used f-strings, because they're awesome, but it requires Python 3.6 (and I think you use 3.5);
2. I used pytest, which I believe is not currently part of your project;
3. I have done some testing (see below), but not extensive testing.

I was wondering how a transformer's voltage regulator should be implemented in GridCal, and came up with this : [https://github.com/miek770/GridCal/commit/e8d8a99c3dadcaa4515362ef18b068be4c6fc473](https://github.com/miek770/GridCal/commit/e8d8a99c3dadcaa4515362ef18b068be4c6fc473). There are a few relativement minor modifications to the Devices.Branch class, as well as a new test file.

Basically, I'm running the full load flow on the grid, and then I check if voltage regulators take action, and if so I run it again.

It appears to work well, but it annoys me a little that this logic is outside of the MultipleCircuit class. I think it's how we currently have to do it because of how the project is structured. It could also be included in a new wrapper function, for ex.: run_pf_with_regulation(grid, options) which would return grid and power_flow.results.

Any thoughts?

SanPen, 2018-10-04:
Hi Michael, 

Excellent work. I have to think about how to integrate this functionality. As it is formulated (using the tap changer objects) I'd implement it as a function inside the class `PowerFlowMP`, along with the PV-PQ switching.  Both the the tap changing and PQ-PV switching logic are discrete operations that belong to the so-called outer loop.

I'll do this as soon as I get some time (this week probably). You're welcome to do it yourself if you need this sooner.

SanPen, 2018-10-04:
I modified the branch to include the two variables that you needed for tap control since that part was trickier (I found a bug on the process too) and I copied the tap control to a function in PowerFlowMP (it does nothing, just copied there)

I confirm that I require python 3.5 or higher no problem with that, and I'll include pytest in the requirements of the setup file.

miek770, 2018-10-04:
Ok thanks.

About Python 3.5, I meant: Must all code be *backward compatible* with version 3.5? If I include f-strings in my code, it won't run under 3.5 and before. If you're developping on 3.6 or 3.7, I would change the minimum version to 3.6 in the README.md to allow f-strings.

SanPen, 2018-10-04:
Ok, no problem, I'll set it to python 3.6

miek770, 2018-10-04:
Great.

I'm working on the pull request, I'm having minor trouble getting it to work with a fork and filtering unwanted stuff. But it works and shouldn't add any significant calculation/compilation if now voltage regulators are used.

miek770, 2018-10-06:
Simply reopened it until it's fully resolved. Will close my PR #24.

miek770, 2018-10-09:
Hi Santiago,

Did you October 5 commits integrate this feature? Is it ready for testing?

Thank you.

SanPen, 2018-10-09:
Hi Michael, 

It's been quite hectic these days.

I have just implemented the feature in a thread safe way, but it is not tested at all.
If you could see if my implementation follows your initial design it'll be nice.

For your reference this is how the compilation works:

`MultiCircuit.ompile()`  ->  Gathers the properties from the objects and creates a `NumericalCircuit` instance.

`NumericalCircuit.compute()` -> Computes the admittance matrix, Sbus, Vbus, computes the islands ans returns a list of calculation-ready islands (instances of `CalculationInputs`)

The instances of `CalculationInputs` are used to run any simulation ensuring that the circuit that the internal matrices and vectores reprensent are non-singular. 

So, if the power flow needs to control the transformer taps, I need to copy the information from the objects all the way to the `CalculationInputs`. This produces a thread-safe object that can be used in parallel computation. It has taken me 3 years to figure this architecture and I have changed GridCal's internal engine 2 times because of this. So this last one seems to be the definitive architecture since it is 13 times faster than the last and it is all vector-oriented.

So, if you need any clarification or find any bug please let me know.

miek770, 2018-10-10:
I sent 3x PR to fix a few bugs that prevented it from working in my test. With these all my tests pass! Thanks for taking the time to integrate this feature, I can see it wasn't exactly easy.

SanPen, 2018-10-11:
BTW, 

I am implementing a storage active power dispatch to control the adjacent branches overloading.

I am not sure how stupid this idea is. Have you ever seen anything similar?

BR,
Santiago

miek770, 2018-10-11:
Hi,

It's certainly not stupid, I currently have a project for a customer who could beneficiate from this, and I expect others to follow as regulation adapts to energy storage. For example, say we have a relatively constant 20 MW generation and a variable 20 MW consumption tied to the same POI (point of interconnection), and very different buying and selling energy costs. Say the buying cost is ~10 times greater than the selling cost.

Ideally, we want the generation and consumption to be always equal, so that no energy transits through the POI. In order to acheive this, an active storage solution could be used as a buffer to "clip" generation to the consumption, and to compensate for the lack of generation when consumption is higher than generation.

That's only one of probably many applications, but it would definitely be a nice feature to incorporate. I could also be used in larger utility studies, i.e. not at the individual facility level. I would probably start by having a look at how [PyPSA](https://github.com/PyPSA/PyPSA) did it, I haven't played with it much but they seem to have already integrated this feature.

They have examples in which they use energy storage to reduce branches overloading at the utility level, as you mentioned. That would be another worthy application.

Are you going to create a new development branch for this?

SanPen, 2018-10-11:
Hi, 

I already have the storage dispatch feature implemented in the optimal power flow in DC and AC approximations. This is how PyPSA does it (I am quite sure) 

The thing is that I am with a real distribution grid from a EU project where the implemented OPF algorithms are not doing as I expect, whereas in IEEE30 it does exactly what I'd expect. So I want a "rule-of-thumb" solution that works without caring about optimality.  In the past I did this by implementing a truncable Nelder-Mead optimization.

I'll include a rough test as a power flow option in the master branch. Since it can be disables (and is by default) there's no need of a new branch.

closed 2018-10-17 by miek770