SanPen/VeraGrid #28: Leakage and magnetizing impedance in doc

miek770, 2018-10-11:
Hi Santiago,

Simple reminder to correct a few terms in section 3.2 "The universal branch model" and 3.3 "The transformer definition from the short circuit test values" in the manual. Zseries is the leakage impedance, and Zshunt is the magnetizing impedance, but both sections present the opposite (i.e. Zseries is said to be the magnetizing impedance, Yshunt/Zshunt the leakage impedance).

Section 3.3.1 "Inverse definition of short-circuit values from the pi model" has an error in equation (3.2):

Zsc = sqrt(R² + 1/X²)

...should be:

Zsc = sqrt(R² + X²)

I would also suggest identifying the pi model as a two-port network pi model. AESO has published a nice reference: (https://www.aeso.ca/assets/linkfiles/4040.002-Rev02-Transformer-Modelling-Guide.pdf)[https://www.aeso.ca/assets/linkfiles/4040.002-Rev02-Transformer-Modelling-Guide.pdf]. Appendix C discusses the two-port network theory and both the T and pi models.

Lastly, regarding the shunt resistance (rfe) / no load losses equivalent conductance (g), the formula presented in Table 2-1 of AESO's doc uses the no-load test MVA, not the nominal MVA. Table 1-1 shows an example where these values greatly differ (25 vs 42 MVA). I will look for other sources, but would you still have yours?

Thanks.

SanPen, 2018-10-12:
Hi, 

Indeed the inverse definition needs to be revisited. I'll make the changes.

SanPen, 2018-10-27:
This is updated in the documentation PDF

closed 2019-01-29 by SanPen