SanPen/VeraGrid #31: New feature: Tolerances

miek770, 2018-10-23:
It would be nice if we could input tolerances for certain values, for example transformers impedance. For example, a power transformer's impedance at the design phase generally varies by +/- 7.5%. For load flow simulations, the higher impedance should be used (i.e. +7.5%). For short circuit simulations, the lower impedance should be used (i.e. -7.5%).

Other typical tolerances in commercial software apply to:
- Reactor impedance;
- Line and cable length.

I'll open another issue to discuss the effect of temperature on cable and line resistance.

SanPen, 2018-10-23:
This would be a nice feature. But we'd need to think the use case very well.
I'd rather have these values change with temperature, Is there a simple model we can follow?

SanPen, 2019-01-29:
What's the status of this one?

miek770, 2019-01-29:
No work done yet, so depending on how you wish to track potential new features, you may close it or tag it accordingly.

It's up to you, if you close it I'll simply reopen it when I'm ready to discuss it further.

SanPen, 2019-01-31:
Honestly I need to understand the use case better.
I guess that the best is to create a python test script "researching" the implementation.
Do you have anything that is worth reading to understand this?

miek770, 2019-01-31:
Hi Santiago,

The official reference would be the IEEE C57.12 series, starting with C57.12.00. It indicates what I mentioned in my first comment: before being tested, there's typically a +/-7.5% tolerance on theoretical transformer impedance.

After that it's mostly common sense, and up to the designer's preferences. I do find some articles that make mention of it but it comes down to: a theoretical power transformer's impedance has a +/-7.5% precision. The designer can choose to ignore it or apply it to be more or less conservative.

If there are sufficient conservative factors elsewhere, the designer may choose to ignore this factor entirely, even during the preliminary stage (i.e. before the transformer is built).

For example, we are contractually bound to meet total system losses below a predetermined value for our wind farm designs, based on our preliminary calculations. In this case, I definitely want to use the most conservative value for the transformers' impedance. For short circuit calculations for equipment duties, I don't care because we already use extremely conservative contributions from the grid. Etc.

Michel

SanPen, 2019-01-31:
So, would it suffice with adding two properties `low_tolerance` and `high_tolerance` to the branch device, and then have some option telling which one is to use?, or is there some more advanced capability required?

miek770, 2019-01-31:
The commercial softwares I use also know which one to use (either low or high), depending on the study (lower impedance for short circuit, higher impedance for power flow), if it's allowed to (user option).

But yes I think it's a very simple enhancement, and I would add a `precision` property to the `Branch` class instead of the 2 properties you suggested (AFAIK it's always ±X%).

It should be an optional property, with a 7.5% default value for transformers, and 0% for other types of branch. The option to consider this precision should obviously be `False` by default.

SanPen, 2019-01-31:
Indeed it looks simple enough, I'll try to implement it as soon as I get some time, or would you rather do it?

miek770, 2019-01-31:
I'd like to finish what I've started with the documentation before starting something else, so I'd be happy either way. Thanks for asking!

SanPen, 2019-02-02:
I implemented the feature, did some playing around and it seems to be doing something, so its ready for a formal test

miek770, 2019-02-07:
I like how you implemented it; it lets the user select the upper / lower band according to his wishes. I'll close this issue, I've added 2 basic tests in #69.

closed 2019-02-07 by miek770