SanPen/VeraGrid #32: New feature: Cable & line temperature

miek770, 2018-10-23:
The resistance of a line or cable varies with its temperature, and it would be a good idea to include this feature to the branch class. Typically, the highest temperature (ex.: 90°C) is used for load flow simulations, and a lower temperature (ex.: 20°C) is used for short circuit simulations.

If the branch model had a base temperature attribute, we could easily add a method to adjust the resistance based on the simulation temperature using the following formulas:

```
R’  = R ( 234.5 + Tc )/( 234.5 + Tb )  Copper Conductors
R’ = R ( 228.1 + Tc )/( 228.1 + Tb )  Aluminum Conductors

where:
    R = Resistance at base temperature Tb
    R’ = Resistance at operating temperature Tc
    Tb = Conductor base temperature in °C
    Tc = Conductor temperature limit in °C
```

This factor may seem neglectable for high voltage transmission networks, but it can be rather significant for certain applications (ex.: for calculating and demonstrating electrical losses vs contractual requirements).

SanPen, 2018-10-23:
This one should be simple enough.

I have always had another formula in the back of my head. 

Formula 4.4 from [here](https://pure.uva.nl/ws/files/2536678/161596_04_1_.pdf)

This was used by consultants back when I worked at KEMA in the Netherlands.

miek770, 2018-10-23:
That's a nice reference, I'll look further into it. We do perform heat dissipation studies for buried cables, either transient or steady-state, but we also sometimes need to simply run a simulation at a certain fixed temperature using the formulas from my 1st post.

I'll work on integrating my other unit tests first, but I could indeed integrate this one as well rather easily when I get a chance. I would probably leave the GUI integration to you though, I admit I've never even tried it (I love the CLI!).

Thanks.

SanPen, 2018-10-26:
Hi, 
I have implemented the simple temperature correction.

The temperature Tc is `temperature` in MultiCircuit
`Tc` and `k` are new branch parameters to model the base temperature and the conductor "thermal constant" (I called it that)

The option to use this correction is implemented as an option in `PowerFlowOptions`

In the GUI the temperature in a double spin box and there is a check to activate the temperature correction.

I have tested it very lightly and it does change the loading indeed. So if you can test it better against ETAP it'll be great.

SanPen, 2018-10-26:
Actually, looking for the formula to change resistance I found the following formula:

`R = R0 * (1 + k * (T0 - T))`

Where `k` can be:

• Copper = 0.004041
• Aluminum = 0.004308
• Iron = 0.005671
• Nickel = 0.005866
• Gold = 0.003715
• Tungsten = 0.004403
• Silver = 0.003819

Being measured in `Ohm/ºC`.

See [here ](https://www.cirris.com/learning-center/general-testing/special-topics/177-temperature-coefficient-of-copper)for the reference.

Do you have any reference for the first formula?

miek770, 2018-10-26:
My reference is closed (in ETAP's help), so not really adequate for an open-source project. I'll ask them what their own reference was and hopefully it'll point to an IEEE standard or publication.

miek770, 2019-01-03:
Closing because of #38 and #53

closed 2019-01-03 by miek770