SanPen/VeraGrid #33: Pulp package not maintained anymore

miek770, 2018-10-24:
Hi Santiago,

I did some research on `pulp` because of the decrepation error it throws when `pytest` is called (see [here](https://pastebin.com/raw/2ZV2eTzZ)) and found that it's no longer maintained: [pulp issues 183](https://github.com/coin-or/pulp/issues/183#issuecomment-432689921).

[thomasnield](https://github.com/thomasnield) suggested moving to [pyomo](https://github.com/Pyomo/pyomo). He also provided a link to a comparison between `pyomo` and `pulp`: [https://pyomocontrib-simplemodel.readthedocs.io/en/latest/knapsack.html](https://pyomocontrib-simplemodel.readthedocs.io/en/latest/knapsack.html).

Should we start looking into migrating from `pulp` to `pyomo`? Based on your experience, would that be very complicated or relatively straightforward?

SanPen, 2018-10-24:
Hi Michael, 

I was aware of Pyomo. It seems that the API is now identical to PULP's.
When I started using PULP, the interface of Pyomo was incredibly convoluted compared to PULP.

Since the two are very similar now, it should be easy to replace pulp with pyomo. But I'd give this a low priority since pulp work fine. I agree though that pyomo is potentially better since it supports non-linear formulations.

miek770, 2018-10-26:
It looks like `pulp` isn't completely inactive, but they're looking for a co-maintainer (see latest response in [pulp issue 183](https://github.com/coin-or/pulp/issues/183#issuecomment-432689921)). I don't plan on getting involved in it (maybe submit a PR to remove the decrepation warning), but moving to `pyomo` may not be needed.

SanPen, 2019-01-29:
They fully rewrote the project. Now GridCal is adapted to the new PuLP.

closed 2019-01-29 by SanPen