SanPen/VeraGrid #41: Some controlled generators not trying hard enough

miek770, 2018-11-05:
Hi Santiago,

I need your help for an issue I'm trying to debug. In my simulations, there are several controlled generators misbehaving. Basically, some generators don't seem to be pushing in the right direction, or not pushing at all. They all have the same 1.0pu voltage setpoint and the same Q limits, but some of them stay at 0pu (Q) and 0.98pu (V) even though their limit is +/-0.02pu (Q).

I did some troubleshooting in [this branch](https://github.com/miek770/GridCal/commit/150fa845806e32e1b1a699e26be243d72aafd83b) (**not for merging**), and it *seems* like `PowerFlowMP.solve()` doesn't care about `vset` for PV buses. The switch logic is fine for switching back-and-forth from PV to PQ, and preventing over/under excitation. But if the bus remains in PV mode, it won't raise or lower its Q in `PowerFlowMP.solve()`.

I really don't know much about power flow algorithms. I might have messed something up in the above branch, but started the investigation because the problem was pre-existing. I could share my Excel grid by email but it'd have to be private, it's for an existing project.

**Brainstorming**

If I understand correctly: the solver tries to solve the matrices to reach the voltage_solution (or V0). If it can, great (it converged). Otherwise, it returns a new voltage solution and tries to converge again based on this new V0.

The initial V0/voltage_solution in GridCal, for PV buses, are the buses' `vset` (more specifically their controlled generators' vset). So the first time it runs, it uses the correct setpoint. But in subsequent iterations it uses the previous result (voltage solution) as its new setpoint, instead of `vset`.

So should we replace the PV buses' voltage_solution be their `vset` after each iteration, if it hasn't reached its Q limit (basically when it remains a PV bus in the switch_logic())?

Thanks.

SanPen, 2018-11-08:
Hi, 

The way the power flow algorithm works is the following:

1 - Solve the voltages that satisfy the numerical problem (Newton-Raphson, Gauss-Seidel, etc...)

2 - Check that the power that is calculated from that voltage solution does not violate the generators Q limits. If some limits are violated do the PV->PQ switching.

3 - Check the transformer taps and adjust them accordingly. 

4 - Go to 1.

Essentially, when a generation bus (PV at first) is switched to PQ, the `vset` does no longer apply because the generator is at either of its Q limits already and can no longer maintain the set voltage. If at a later iteration the calculated Q of the switched bus is within limits, the bus is switched back from PQ to PV.

The PV-PQ switching is a good trick, but It does not mean that there are no other ways to solve the problem. One could throw the problem to a convex solver and expect it to solve the voltages taking into account the Q limits and any other limit that there might be. But that is impractical and could potentially lead to no solution at all. Not to mention that it might not resemble to the actual physical process in the grid.

There might be better heuristic algorithms to get the generators limits included but I haven't heard of any other better than PQ-PV switching, specially on terms of speed.

miek770, 2018-11-08:
Hi Santiago,

Thank your for the explaination. To follow-up on your email:

> The first thing I notice is that you choose the LM (Levenberg-Marquardt) solver. If I run a simulation with it it does not converge. However if I set the solver to NR (Newton-Raphson) everything works fine.
> 
> Please try and see if this solves the problem.

I was going to write: "Strangely, it does converge for me with the LM (tried on 3 machines, win7, win10 & linux). I tried the NR, and got the same results (ex.: M95 has a vset of 1.0, Qlimits of +/- 2 MVAr, settles at 0.977 pu and Q = 0.24 pu)."

...but then I tried the GUI (I always run GridCal from the CLI or scripts) and LM doesn't converge, even with 99 max iterations. In the CLI it does converge. I also get very different results from the NR method in the GUI vs the CLI. I'll compare how the GUI and my script call the execution of the load flow and see if I can find why both behave so differently.

SanPen, 2018-11-08:
After checking, I see that the problem is in the function `get_report_dataframe`. 

Both with the script and the GUI I get the same complete convergence report:

converged[True, False, True, True, True, True, True, True, True, True, True, True]

        inner_iterations: [4, 20, 3, 3, 2, 2, 2, 2, 2, 1, 1, 1]

        outer_iterations: 12

        error: [5.142464110733989e-08, 493060286590706.8, 2.2326737706180275e-09, 8.331796476450647e-11, 3.9991814009197996e-05, 2.787779027009459e-07, 6.172819297378673e-09, 7.832294621208686e-09, 1.2299288224204776e-08, 8.474123339121883e-05, 4.56366587081973e-05, 9.937326288633058e-05]

        elapsed: [0.029719829559326172, 0.14361023902893066, 0.020978212356567383, 0.02116107940673828, 0.014257192611694336, 0.014130592346191406, 0.01414346694946289, 0.014140129089355469, 0.014181852340698242, 0.0071103572845458984, 0.007142543792724609, 0.0071258544921875]

        methods: [<SolverType.NR: 1>, <SolverType.NR: 1>, <SolverType.IWAMOTO: (8,)>, <SolverType.IWAMOTO: (8,)>, <SolverType.IWAMOTO: (8,)>, <SolverType.IWAMOTO: (8,)>, <SolverType.IWAMOTO: (8,)>, <SolverType.IWAMOTO: (8,)>, <SolverType.IWAMOTO: (8,)>, <SolverType.IWAMOTO: (8,)>, <SolverType.IWAMOTO: (8,)>, <SolverType.IWAMOTO: (8,)>]

The `get_report_dataframe` function is only getting the first values for one island and that mislead me. Indeed there's a lot of PQ-PV switching, and that is normal for the grid you provided.

I'll fix the function. But apart from it, are you expecting another behaviour from the power flow routine?

miek770, 2018-11-08:
Great thanks.

I'm quite sure that almost all the PV generators should be able to meet the desired voltage (1.0 pu). The scenario I'm running is far from being excessive (much more difficult scenarios will need to be done afterwards to test the windfarm's limits). So I'm expecting all of them to be at V=1.0 pu at the end of the simulation. Most of them aren't.

If they fail somehow, I'd expect any generator that fails to be at either of its Q limits, not somewhere in the middle.

SanPen, 2018-11-08:
If I run a single iteration with no controls on, this is the reactive power in the PV nodes in MVAr:
```
-6,  0,  6,  2, -2, -4,  4, 15,  7,  4, -3, -5, 18,  9, -2,  1,  4, 13, 12, 25,  0, 14, 41, 63,  0, -2, 13, 28, 16, 11, -6, -2,  1, 65,  7, 17, 27, 16,  7,  1,  0, -4,  0,  4, 14, 73, 38, 51
```
So yes, almost all violate the +-2 MVAr limit by a lot. I don't think that there is much to be done here. The PQ-PV switching does its job and manages to converge after 12 outer loop iterations.

In this particular grid, this means that to maintain v=1 p.u. "everywhere", there is a huge amount of reactive power exchange, maybe an impossible or unrealistic amount of reactive power. But this is more a philosophical / engineering discussion I think.

miek770, 2018-11-08:
Maybe we're simply starting too far from the final solution? I think it would be impractical to have to manually provide an approximation of the final for each scenario, but do you think that would help?

This definitely works in real life and in other softwares (ex.: ETAP), we've designed and studied a lot of wind farms in the past. I could make this scenario "simpler", but as I said I'll have to run much more difficult scenarios aftewards and it will become increasingly difficult to guess the final solution before running it.

SanPen, 2018-11-08:
The power flow problem presents a convex behaviour in the vicinity of the solution, meaning, if most buses are voltage controlled and we initialize the power flor program with those voltage values, we're pretty close to the solution. This means that the solution is what it is. To bring the reactive power within limits you may have to change the active power or change the grid.

The only way to know if the problem is feasible at all is tu run an optimal power flow linearising the equations like in the AC-OPF implementes, but adding the active and reactive power constraints. Then if this problem is infeasible there's no solution since the simplex algorithms guarantee the global optimal.

miek770, 2018-11-09:
I spent a few hours trying to solve this, ouch it's not exactly easy. Tried changing the switch logic entirely, keeping Vset_initial and doing some tricks, but it doesn't work. The issue is that all generators affect several others and since a lot changes every iteration, it just doesn't manage to settle on a valid solution (even if it converges). For example if one generator goes from 41 MVAr to 2 MVAr in a single iteration, it will significantly lower the voltages of this common branch. In other words, it converges to an unstable solution.

Maybe if I found a way to do it more incrementally... by doing smaller corrections each time. I wish OTI would tell me how they do it in ETAP (it works)...

What I'm not entirely sure I understood during my tests was that if I messed up with Vnew (always using `Vnew[i] = complex(module, 0)`), and did 999 iterations (`control_max_iter = 999`), the active power would get messed up. If I understand correctly, it's related to what you mentioned in your previous message:

> This means that the solution is what it is. To bring the reactive power within limits you may have to change the active power or change the grid.

Because I'm trying to force the solution to what I want, by changing the voltages, it does what it needs to do (i.e. adapt reactive **& active** power) to converge to the requested voltages. Is that correct?

Anyways, I'll keep thinking about it and look for a solution; I need this to work. Even if it's an ugly hack that increase simulation time.

miek770, 2018-11-09:
Hum I could simply do it exactly like the transformer voltage regulators: set them to PQ and incrementally change their Q until I reach the desired V. I'll try this approach and report.

SanPen, 2018-11-10:
I would suggest to do a literature review for better methods. If we check IEEE papers there might me a method simmilar to what ETAP does. This is how I foynd the currently implemented method, but there might be better ones.

miek770, 2018-11-10:
It works! I added an iterative_pv_control method as an option, and it works! See here: [https://github.com/miek770/GridCal/commit/dccc03845518404398e572a1f567caee1616eed7](https://github.com/miek770/GridCal/commit/dccc03845518404398e572a1f567caee1616eed7)

It doesn't affect normal behavior, switch_logic is applied by default. But it solves my network in 753 iterations (outer loop). I can reduce the number of iterations by tweaking `precision` and `increment` but then the quality of the results decreases. It should probably also be options, but I'm guessing that simpler grids will be solved with fewer iterations anyways.

Another method would be to make the increment variable, by making of rule-of-thumb for the relation between the Q change and the expected impact on V. But that's for another day!

So I agree with your last post; more research could and should be done to simply improve the base method in the long run, but would you be fine with this being merged to GridCal as it is (or with adjustments)?

miek770, 2018-11-10:
Ok reopening the issue, I was a bit quick to close it.

SanPen, 2018-11-11:
I propose to you to have an enumeration in the options to determine the Q-control mode. This makes the options easier to handle. Something like None, Direct and Iterative.

miek770, 2018-11-11:
Good idea, plus there might be other methods in the future to add to the list, and it reduces the number of required options.

miek770, 2018-11-12:
I believe all that's left before we close the issue is the integration to the GUI?

A last note about PR #44: If we ever find the steepness factor "k" to be to high for special cases, we could adapt the factor after the first iteration, based on the impact of the Q increment on the voltage magnitude. I think it would be a more elegant solution than adding another option to let the user select "k" manually.

In other words, **if we ever discover that a static "k" can lead to instability**, we could automatically calculate the dV/dQ ratio for each generator and compensate "k" accordingly.

SanPen, 2018-11-12:
The GUI should be able to select the Q-Control mode from a drop down menu, so that should be it.
Indeed the k can stay as it is. However the maximum outer loop iterations should be an option. I'll take care of that.

miek770, 2018-11-12:
I saw your commit, looks good. Closing the issue.

closed 2018-11-12 by miek770