SanPen/VeraGrid #59: mpi4py

miek770, 2019-01-20:
Hi Santiago,

The new dependency to `mpi4py` might be a bit problematic in some cases; it requires additional installation steps which are very platform dependant, and it's preventing Read the Docs from running autodoc (see #58 and [this failed RTD build](https://readthedocs.org/projects/gridcal-wip/builds/8422123/)).

Could we make this dependency optional? As it stands, I believe it breaks the simple installation procedures shown in the README.

Thanks,

Michel

SanPen, 2019-01-21:
Hi, 

I don't think mpi4py is being used in "production" code. It might be a dependency of PySOT. Even the PySOT dependent code is not accesible from the UI. 

But, from what I know about mpi4py, you need to launch the python program with `mpi4py python script.py` or a similar command.

I cannot say that not installing mpi4py would make GridCal fail (not the installation, which apparently does...), this needs some attention.

miek770, 2019-01-21:
Ok I'll try to reproduce, but on my windows machine I couldn't even run the pytest code because of mpi4py. I had to install Microsoft MPI, add it to the system's path, and then`pip install mpi4py`.

On Linux I can't remember having had to install anything particular... But I might simply have forgotten because it's so straightforward (`sudo apt install <mpi-package>`).

Michel

SanPen, 2019-01-29:
On Windows 10, I've been unable to reproduce a situation where mpi4py does not install.
I installed GridCal and I did not see mpi4py as a dependency.

miek770, 2019-01-31:
#61 fixes it all: It removes `mpi4py` from required dependencies and prevents trying to import `mpiserve` if `mpi4py` is not installed). Closing.

Otherwise, running `pytest` without `mpi4py` installed resulted in:

```
$ pytest
============================= test session starts =============================
platform win32 -- Python 3.6.6, pytest-3.8.1, py-1.6.0, pluggy-0.7.1
rootdir: C:\Users\michel\gits\windfarm\GridCal, inifile: pytest.ini
collected 12 items / 1 errors

=================================== ERRORS ====================================
_______________ ERROR collecting src/tests/test_line_losses.py ________________
ImportError while importing test module 'C:\Users\michel\gits\windfarm\GridCal\src\tests\test_line_losses.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
src\tests\test_line_losses.py:1: in <module>
    from GridCal.Engine.Drivers.power_flow_driver import PowerFlowOptions, PowerFlow
src\GridCal\Engine\__init__.py:5: in <module>
    from GridCal.Engine.Replacements import *
src\GridCal\Engine\Replacements\__init__.py:1: in <module>
    from GridCal.Engine.Replacements.mpiserve import *
src\GridCal\Engine\Replacements\mpiserve.py:18: in <module>
    from mpi4py import MPI
E   ModuleNotFoundError: No module named 'mpi4py'
============================== warnings summary ===============================
c:\users\michel\gits\windfarm\lib\site-packages\pulp\solvers.py:71: DeprecationWarning: The SafeConfigParser class has been renamed to ConfigParser in Python 3.2. This alias will be removed in future versions. Use ConfigParser directly instead.
  'os':operating_system, 'arch':arch})

-- Docs: https://docs.pytest.org/en/latest/warnings.html
!!!!!!!!!!!!!!!!!!! Interrupted: 1 errors during collection !!!!!!!!!!!!!!!!!!!
===================== 1 warnings, 1 error in 1.91 seconds =====================```

closed 2019-01-31 by miek770