SanPen/VeraGrid #60: Docstring format

SanPen, 2019-01-21:
Hi Michel ( @miek770 ),

I see that in the branch device there is another style of documenattion.
Why did you choose that format?

Before I was using the style that is automatically generated by pycharm. I agree that the style was not consistent because for some reason the default pycharm format varies.  At least for me it is important that an IDE can generate the docstring automatically from a function definition.

BR,
Santiago

miek770, 2019-01-21:
Good morning,

In the future please don't merge PR with "WiP" in the title, #58 wasn't completely ready. There was an explanation in it: the new proposed docstring format is in reStructuredText to be compatible with autodoc/Sphinx.

That being the most common and official Python docstring format, as far as I know, pycharm should be able to recognize it. See [here](https://www.jetbrains.com/help/pycharm/documenting-source-code.html), although I didn't test it (I use either good old vim on Linux and Geany on Windows).

Thanks,

Michel

miek770, 2019-01-21:
> In the future please don't merge PR with "WiP" in the title, #58 wasn't completely ready.

Actually, this might be a good indication that a contribution guide might be useful. Something like this: [https://gitlab.com/miek770/energy_tools/blob/master/CONTRIBUTING.md](https://gitlab.com/miek770/energy_tools/blob/master/CONTRIBUTING.md). I shamelessly copied and adapted pandapower's guide for this small project.

How would you like contributors to present branches that are not ready to be merged, but ready for discussion and feedback? I think the "Work in progress / WiP" prefix is widespread, but I could also have done it in an issue an link to my own branch.

Michel

SanPen, 2019-01-29:
Actually I don't know if you can have a github branch without a pull request appearing (I guess so)

I'd propose that only final pull requests are made if that is possible. That way I can be more effective.

miek770, 2019-01-29:
Noted. I won't create PR until they're ready to be merge (IMO), and will include something to this effect in the doc. I can create a branch in my fork and link to it in an issue in the main repo, without creating a PR.

About the docstring format, are you ok with reStructuredText? Did you manage to set pycharm accordingly?

miek770, 2019-02-02:
Great, I think this issue could be closed if that's ok with you.

closed 2019-02-02 by SanPen