SanPen/VeraGrid #62: Read the docs

miek770, 2019-02-02:
Hi Santiago,

So the new documentation is pretty much ready to go, I'd like to have your input before doing the PR. Could you please check it out here: [https://gridcal-wip.readthedocs.io/](https://gridcal-wip.readthedocs.io/)?

Please note that I've adapter the README.md accordingly, see here: [https://github.com/miek770/GridCal/blob/doc/README.md](https://github.com/miek770/GridCal/blob/doc/README.md). I've also added a basic contribution guide here: [https://gridcal-wip.readthedocs.io/development.html#contributing](https://gridcal-wip.readthedocs.io/development.html#contributing), following #60.

Next steps:

- Remove the GitHub wiki, and integrate its information to Read the Docs (RTD);
- Expand autodoc's coverage to additional modules/classes (reformat docstrings accordingly);
- Integrate the HVDC manual;
- Remove GridCal's PDF manual from the repo;
- Activate PDF generation on RTD.

Let me know what you think!

Michel

andreas1024, 2019-02-04:
Hi, I just read through your "Newton-Raphson-Iwamoto" page - looks quite good - however, I noticed some missing line breaks between equations and I might be wrong, but I think the two derivatives of S near the bottom have their right sides interchanged.

miek770, 2019-02-05:
Thanks @andreas1024, I've created PR #66 for the missing line breaks. I'll update the docs once it's merged (having some issues with RTD builds failing unless i manually wipe the previous build each time).

@SanPen: Any thoughts on the other issue (about the 2 derivatives of S)?

closed 2019-02-25 by SanPen