SanPen/VeraGrid #70: Newton-Raphson doc problem

miek770, 2019-02-07:
Hi Santiago,

This commit: https://github.com/SanPen/GridCal/commit/8f6e4dc3a490b4be3e59dbd0656e764b2a069a41

...introduced some bugs in the documentation, see here: https://gridcal.readthedocs.io/en/latest/power_flow/newton_raphson.html

Michel

SanPen, 2019-02-07:
I'll have a look

Is there a way I can compile this and review it?

In pycharm it did not crashed

miek770, 2019-02-07:
Did you try to `make html` it beforehand? Otherwise I use a "WiP" RTD site to test it, you could also setup one. Or simply create a RTD account and I'll invite you to the official one (so you can investigate if it fails).

miek770, 2019-02-08:
Hi Santiago,

I took the liberty of going through the NR doc and propose corrections, see #71. The key points are:

1. Check if `./make.bat clean ; ./make.bat html` threw any error (it did);
2. Added a bunch of links between sections;
3. Made all titles, apart from the main one, subtitles (underscored by "-" instead of "=");
4. Added a blank space after all `.. math::` lines (required);
5. Removed all `\begin{equation}` and `\end{equation}`;
6. Split different equations with individual `.. math::` statements to get a nice center alignment (otherwise grouped equations get right-aligned by default, weird default behavior...);
7. Replaced refs in LaTeX format with Sphinx' format;
8. *Try* to stick to a line length of 88 characters.

With these changes `./make.bat html` no longer throws errors. My *WiP* doc also looks fine: [https://gridcal-wip.readthedocs.io/en/fix_nr_doc/power_flow/newton_raphson.html](https://gridcal-wip.readthedocs.io/en/fix_nr_doc/power_flow/newton_raphson.html).

Hope this helps, let me know if you have questions.

Michel

SanPen, 2019-02-25:
This one is solved too, correct?

miek770, 2019-02-25:
Indeed, the 8 points above were there to help (for your information).

closed 2019-02-25 by miek770