SanPen/VeraGrid #72: Create webhook for RTD

miek770, 2019-02-09:
Hi Santiago,

I couldn't create a webhook on read the docs for this repo because it's on your account. The webhook is used to automatically update the doc each time there's a commit in the followed branches (**master** in this case).

The documentation to setup a webhook on GitHub are here: [https://docs.readthedocs.io/en/latest/webhooks.html](https://docs.readthedocs.io/en/latest/webhooks.html).

Otherwise I (or you if you create a RTD account) have to do it manually each time. I just updated then for yesterday's merged PR.

Sorry, I would have liked to setup it completely but I'm stuck.

Michel

SanPen, 2019-06-11:
Hi Michel, 

Is this something we can fix now I have a RDT account?

Santiago

miek770, 2019-06-11:
Yes, but I believe you have to do it, you'll need to use your Github credentials somewhere (the link above should provide the relevant instructions, I had no trouble creating the webhook for my fork).

SanPen, 2019-06-12:
Done!

closed 2019-06-12 by SanPen