SanPen/VeraGrid #76: Power flow heuristics documentation

SanPen, 2019-02-12:
@miek770 , it'll be great is you could document the heuristic algorithms made for tap adjustment and reactive power control. 

No rush, but just to keep track.

BTW, I'm thinking about using some project management tool to make a roadmap and have tasks like [Zenkit](https://zenkit.com)

miek770, 2019-02-12:
Indeed, I'll document these in the near future.

Good idea for Zenkit (or some other similar tool). This is something I haven't used before but which could be very useful for many of my projects, including this one.

miek770, 2019-02-17:
I think #80 completes this one. Let me know if you'd like additional details or if you have any comment!

Michel

closed 2019-02-21 by SanPen