SanPen/VeraGrid #84: Make Q iter control k factor adjustable

miek770, 2019-02-24:
Hi Santiago,

I'm working on [this branch](https://github.com/SanPen/GridCal/compare/master...miek770:q_control_adjustable_k?expand=1) to make the iterative Q control method's steepness factor `k` adjustable through a *PowerFlowOptions*, but I'm getting errors about missing icons when regenerating the UI:

```
michel@DESKTOP-ERT4IKV MINGW64 ~/gits/windfarm/GridCal/src/GridCal/Gui/Main (q_control_adjustable_k)
$ python update_gui_file.py
Cannot find file: icons/check_all.svg
Cannot find file: icons/uncheck_all.svg
Cannot find file: icons/copy2down.svg
Cannot find file: icons/next.svg
Cannot find file: icons/prev.svg
Cannot find file: icons/color_grid.svg
```

Were they removed purposefully?

Otherwise it seems fine; the options works both in the GUI and the CLI.

See doc here: [https://gridcal-wip.readthedocs.io/en/q_control_adjustable_k/api/drivers/pf_driver.html#pf-options](https://gridcal-wip.readthedocs.io/en/q_control_adjustable_k/api/drivers/pf_driver.html#pf-options)

Thanks,

Michel

SanPen, 2019-02-24:
Hi, I've just added the missing icons

miek770, 2019-02-24:
Thanks, integrated to #87

closed 2019-02-24 by miek770