SanPen/VeraGrid #86: Automatic precision fails in GUI

miek770, 2019-02-24:
Hi Santiago,

When running a PF in the GUI with "Automatic precision" checked, I get the following error:

```
Traceback (most recent call last):
  File "C:\Users\michel\gits\windfarm\GridCal\src\GridCal\Gui\Main\GridCalMain.py", line 1830, in run_power_flow
    lg = np.log10(abs(self.circuit.numerical_circuit.Sbus.real))
AttributeError: 'NoneType' object has no attribute 'Sbus'
```

...which I assume happens because MultiCircuit hasn't been compiled yet. If I run a PF without this option first, and then run it again with this option, I get:

```
Traceback (most recent call last):
  File "C:\Users\michel\gits\windfarm\GridCal\src\GridCal\Gui\Main\GridCalMain.py", line 1830, in run_power_flow
    lg = np.log10(abs(self.circuit.numerical_circuit.Sbus.real))
AttributeError: 'NumericalCircuit' object has no attribute 'Sbus'
```

I can see that NumericalCircuit doesn't have an `Sbus` attribute, but I'm not sure what the value we want to use is exactly.

Regards,

Michel

SanPen, 2019-02-24:
Fixed!
That syntax is very old, now I updated it to the new calculation objects

closed 2019-02-24 by SanPen

miek770, 2019-02-24:
Awesome, thanks!