SanPen/VeraGrid #88: Doc rendering issues

miek770, 2019-02-25:
Hi Santiago,

There are a few rendering issues with this section: [https://gridcal.readthedocs.io/en/latest/models/from_objects_to_matrices.html](https://gridcal.readthedocs.io/en/latest/models/from_objects_to_matrices.html).

For the right-justified formulas, you have to break them into individual `:math:` sections to get them all to be centered.

For the figures, I haven't looked into it but you could compare with another section that has some to find the difference.

Good evening,

Michel

SanPen, 2019-02-25:
Should be fixed, but please check.

miek770, 2019-02-25:
Sorry but no, images still don't show up (see here: [https://gridcal.readthedocs.io/en/latest/models/from_objects_to_matrices.html](https://gridcal.readthedocs.io/en/latest/models/from_objects_to_matrices.html)).

SanPen, 2019-02-26:
How do you upload the generated documentation from `build_` into the website?
I did generate the docs, and the pictures are there, I only don't know how to upload them to the website.

miek770, 2019-02-26:
You don't, ReadTheDocs recompiles it on its own virtual environment. The build folder is ignored by git (as it should, see [https://github.com/SanPen/GridCal/blob/master/.gitignore](https://github.com/SanPen/GridCal/blob/master/.gitignore)).

See the build output here: [https://readthedocs.org/projects/gridcal/builds/8653614/](https://readthedocs.org/projects/gridcal/builds/8653614/). You can expand the individual lines, it shows that several images are imported:

```
copying images... [ 16%] power_flow/../figures/matrix-reduction.png
copying images... [ 33%] models/../figures/ElementBusAgregation.png
copying images... [ 50%] figures/gui_ieee_39.png
copying images... [ 66%] models/../figures/pi_trafo.png
copying images... [ 83%] power_flow/../figures/coefficients_structure.png
copying images... [100%] models/../figures/CompilationProcess.png
```

I think they should all be in `figures/image_file.ext`, not in a path including `..`. That might be why they don't appear.

hope this helps.

miek770, 2019-02-27:
It's ok now: https://gridcal.readthedocs.io/en/latest/models/from_objects_to_matrices.html

I think I forgot to regenerate the doc after your fix, sorry about that. I'm a bit swamped these days. Closing.

closed 2019-02-27 by miek770