SanPen/VeraGrid #92: Customize control algorithm for time-base simulation

vermouth1992, 2019-03-19:
Thanks for the great work. I just wonder is it possible to perform customized control algorithm in this framework. Something like
```
circuit = build_circuit()
while controlstep < maximum_steps:
    observation = circuit.sample_values()
    circuit.do_customized_control(observation)
    controlstep += 1
```

For example, I would like to control capacitor bank such that the voltage at a certain node of the circuit is maintained within a bound.

SanPen, 2019-03-19:
Absolutely possible. @miek770 has done something very similar to control the voltage using taps and the reactive power.

The file with the important routines is: [this one](https://github.com/SanPen/GridCal/blob/master/src/GridCal/Engine/Drivers/power_flow_driver.py)
And the important function to look at is `single_power_flow` in the line 524.

I guess that you may want to build some external proof of concept first. I you need help please tell us.

SanPen, 2019-03-25:
@vermouth1992 Any feedback?

Otherwise I'll close the issue (you may comment any time though)

closed 2019-03-25 by vermouth1992