SanPen/VeraGrid #93: "Load all the grids" test

miek770, 2019-05-25:
Hi @SanPen,

There are a few warnings when running `pytest` which all originate from `test_load_all_the_grids.py`, here's the output: [https://pastebin.com/raw/j4cqJjXj](https://pastebin.com/raw/j4cqJjXj)

I didn't investigate further, but it looks like the grids haven't been adapted to the latest changes in the TransformerType class.

Michel

SanPen, 2019-05-26:
I could not reproduce the problem because I was getting a Qt related error.
I'm on ubuntu 18.04 and the error was about PySide2 Qt version  not matching the system version...

So what I did was to download the python sources, compile them and install GridCal so I get all the dependencies there, then uninstal the GridCal package and use that newly compiled environment to develop.

Then the tests passed with no problem.

I'd recommend by no means to compile python under windows, but rather install a fresh installation and not to include it in the system PATH variable during the installation. Then install GridCal's dependencies and develop with that one.

I think that PyQt5 "pollutes" the python environment so it is either PySide2 or PyQt5.

Funny enough, GridCal runs fine in the "polluted" system python3, but the tests failed because of PySide2.

Tell me if you face any other issue.

miek770, 2019-05-31:
Ok I'll try other installation methods, but this was a fresh installation of the `master` branch in a new virtual environment (`python -m venv`). I'll let you know if the problem stays or disappears.

miek770, 2019-06-09:
Hi,

It was an issue with example_transformer_tpe.xlsx, which didn't refer to the correct headers for the transformer type. Basically:

hv_nominal_voltage > HV
lv_nominal_voltage > LV
nominal_power > rating
copper_losses > Pcu
iron_losses > Pfe
no_load_current > IO
short_circuit_voltage > Vsc

Can be closed once #98 is merged.

closed 2019-06-10 by miek770