SanPen/VeraGrid #100: Add Continuous Integration

Bengt, 2019-06-24:
There is now a proper test suite which can be used to test the code for regressions. With minor modifications, these tests can be run in a continuous integration environment. This enables them to be executed on every push, thus catching regressions before publication of new docs and package versions. Catching errors after pushing and before releasing can prevent some errors to affect users. To demonstrate the feasibility, I set a pipeline up over on GitLab.com, because that is what I know my way around best:

https://gitlab.com/gridcal/gridcal/pipelines/67806167

SanPen, 2019-06-25:
This is a very good suggestion.

Honestly I have no idea how to integrate automated tests into GitHub. So I'd ask you please to help me here and do some research on the steps and such. That would help me to get it done sooner.

Best regards,
Santiago

Bengt, 2019-06-25:
Hi Santiago,

thanks for the quick reply. I can totally understand your confusion. GitHub requires repository owners to integrate third-party services for continuous integration, which is quite involved especially when doing that for the first time. I used multiple (Drone,io Travis.ci etc.) external continuous integration services in past GitHub projects. The flexibility this brings is great for advanced users and big projects with many contributors. However, if one just needs to run the tests regularily, simpler solutions would already suffice. Since the GitLab mirror is already running as CI service for an external repository, the easiest way forward would be to make you maintainer in the project over there. That would enable you to administer the project and group yourself. I would offer to help you with that, since I am now quite an intensive user of GitLab and thus know my way around a bit there.

Regards,
Bengt

miek770, 2019-06-30:
It's a great idea, and would be a good motivator to create more tests. But switching between GitHub and Gitlab seems a lot more complicated than setting Travis up on GitHub. See these simple steps: https://www.smartfile.com/blog/testing-python-with-travis-ci/

I haven't tested them myself, and I can't make commitments these days, but it looks really simple.

P.s.: travis-ci.org is free for open source projects.

P.p.s.: I haven't compared it to other alternatives, but it's free and simple to setup, so if it works well....

SanPen, 2019-06-30:
Hi!

I added travis-CI automated build and it seems to be working.

However the builds fail due to PySide / Matplotlib related stuff ([see my question on stack overflow](https://stackoverflow.com/questions/56826665/correct-way-of-setting-pyside2-as-matplotlib-backend))

We'll see if it gets solved.

PR #101 (2019-07-01): fixed travis-ci
Fixed #100 : You have to enable the xvfb service: https://benlimmer.com/2019/01/14/travis-ci-xvfb/ to enable the test with libraries that a graphic environment needs.

Result: [image] :smile:
closed 2019-07-01 by SanPen