SanPen/VeraGrid #102: matplotlib 3.1.0 fails on readthedocs

miek770, 2019-07-01:
Hi Santiago,

The last few builds on readthedocs failed because it looks like it can't install the latest version of matplotlib (3.1.0), see here: [https://readthedocs.org/projects/gridcal/builds/9306186/](https://readthedocs.org/projects/gridcal/builds/9306186/)

Did you need this particular version or could you stay with 3.0.3 for the time being? [https://pypi.org/project/matplotlib/#history](https://pypi.org/project/matplotlib/#history)

I made a branch with version 3.0.3 and the doc builds (see [https://readthedocs.org/projects/gridcal-wip/builds/9307871/](https://readthedocs.org/projects/gridcal-wip/builds/9307871/)). If you're fine with version 3.0.3, you can create a PR from here: [https://github.com/miek770/GridCal/tree/matplotlib_3_0_3](https://github.com/miek770/GridCal/tree/matplotlib_3_0_3)

P.S.: Do you receive email warnings when a RTD build fails? If not, you can enable it from your portal.

Bye,

Michel

SanPen, 2019-07-01:
I reverted back to 2.1 for me it doesn't matter but I guess that the Qt backend will force us to have a 3.x.x version of matplotlib but I am not sure which one.

miek770, 2019-07-02:
Ok, 3.0.3 would probably be safer than 2.1 but as long as there's no problem... it's all good, thanks.

closed 2019-07-02 by miek770