SanPen/VeraGrid #104: Unable to run GridCal and Spyder simultaneously - Abort error in GridCal

rosadoalberdic, 2019-07-22:
I installed GridCal by following the instructions in the web and typing in the console "pip install GridCal". When trying to run GridCal in the Anaconda Prompt with the following command: python -c "from GridCal.ExecuteGridCal import run; run()", I get the following error:

>_ _qt.qpa.plugin:_ Could not load the Qt platform plugin "windows" in "" even though it was found. This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem. Available platform plugins are: direct2d, minimal, offscreen, windows.

I read that this can be due to the qt.conf file which is inside of the Anaconda installation folder. When deleting this file, I can run GridCal successfully, but I cannot open Spyder anymore and the following message appears when trying to do so: 
[image]

So it seems that Spyder cannot be run without that qt.conf file. I am not sure whether this is related to the same problem, but when typing anything in the GridCal console, such as importing any packages, the "ERROR: execution aborted" message appears. 

I would appreciate if someone could give me a hint of what is happening here since I need to work simultaneously in Spyder and GridCal, but I cannot have both open for the reasons mentioned above.

Thank you very much in advance!

Best regards,

Cristina

SanPen, 2019-07-22:
Hi, 

Indeed it is a problem. As far as I can tell there is no easy solution with Anaconda.

I can propose you two solutions:

- Avoid Anaconda and install python from python.com and install the packages by yourself.
- Download the standalone [GridCal setup for windows](https://sanpv.files.wordpress.com/2019/05/gridcalsetup-1.zip), then install  and then update. This will provide GridCal with its own python distribution that I compiled myself.

Please let me know if it works for you.

SanPen, 2019-07-22:
A side note here (not a solution) would be to use [pycharm](https://www.jetbrains.com/pycharm/download/download-thanks.html?platform=windows&code=PCC) to develop your work instead of Spyder.

At the beginning it might be harder to use, but the learning curve pays.

In the mean time I will investigate a real solution.

SanPen, 2019-07-28:
So, after investigating I have concluded that it is not possible to run both GridCal and Spyder in anaconda.

However it is totally possible and drama-free using a normal python setup (from python.com) and installing GridCal and Spyder with pip.

I hope that the Anaconda guys step up their game and support the Official Qt library for python (PySide2)

closed 2019-08-20 by SanPen