SanPen/VeraGrid #105: Reintroduce Semantic Versioning

Bengt, 2019-07-25:
In order to upgrade GridCal reliably, one needs to be able to tell which version is greater by the version number. GridCal has used semantic version numbers for this purpose for almost a year (3.0.1 - Sep 27, 2018) now. However, GridCal 3.5 breaks this feature by dropping the patch level from the version number. Now it is unclear whether version 3.5 is larger or smaller than version (non-existent) 0.3.6. I would suggest adding the patch level back in again, maybe in a version 3.5.1.

SanPen, 2019-07-25:
totally forgot about it!
My intention is definitely to keep 3 numbers

Bengt, 2019-07-25:
Hi Santiago,

thanks for the quick reply! I recently ran into this kind of problem, with another library, so prefered to report it, before anyone gets similar issues with GridCal.

Regards,
Bengt

SanPen, 2019-07-28:
Hi, 

I have just uploaded 3.5.1 to Pypi, so it should be fixed.

Best regards,
Santiago

Bengt, 2019-07-28:
I can confirm. My project's pipeline runs fine with GridCal pinned to 3.5.1:

https://gitlab.com/pfasdr/code/main/pipelines/73348504
(Should be a private URL, so just for my future reference)

Bengt, 2019-07-28:
Thanks for the fix!

closed 2019-07-28 by Bengt