SanPen/VeraGrid #107: Consider HELMpy

Bengt, 2019-07-28:
Hi Santiago,

I want to point you towards this HELM implementation I just found:

<https://github.com/HELMpy/HELMpy>

I thought it would be nice to link these two projects from each others README file, to increase each other's visibility.

The API of that implementation seems very cumbersome. It reads matlab and excel files and writes results to a txt file or something. So I guess integrating it into GridCal is not an option at the moment. Does the implementation look interesting to you maybe?

Regards,
Bengt

SanPen, 2019-07-28:
I'll have a look at what they've done.

I also have just opened an issue inviting them to integrate with GridCal...we'll see.

Bengt, 2019-07-28:
Let's track progress there then.

https://github.com/HELMpy/HELMpy/issues/3

closed 2019-07-28 by Bengt

SanPen, 2020-02-25:
Hi, I received a miracle contribution with a HELM formulation that actually works
You can check it out