SanPen/VeraGrid #108: Hide warnings

miek770, 2019-08-05:
Hi @SanPen,

I'm currently using GridCal with [pyswarms](https://pyswarms.readthedocs.io/en/latest/index.html) to research automated optimal grid design. The first step implies running millions of power flow simulations (ACLPF), several of which having no slack node. I need to shut the warnings off. I thought about simply calling the power flow this way from my code:

```
with warnings.catch_warnings():
    power_flow.run()
```

...but that doesn't work; warnings still show in the terminal. [warnings.catch_warnings()](https://docs.python.org/3/library/warnings.html#warnings.catch_warnings) is described as being not thread-safe, so that might be why it fails to hide warnings in this context.

So, I'd like to add a **quiet** flag to `PowerFlowOptions`, with either a temporary filter, or a bunch of checks in the code:

```
if not quiet:
    warn("Warning message")
```

I can prepare a pull request but I'd like to know your preference. Ideally, the same behavior would be implemented in the other modules as well.

Thanks!

SanPen, 2019-08-05:
Hi!

I have just removed all the warnings from the Power flow driver.
I realised this was my idea already since I had a logger list already there.

So the way it works now is that when a power flow is finished, the programmer should check the logger list for messages. I do that from the GUI and no console warning is displayed from the power flow module.

So I guess you don't need to do anything but update the code.

Please let me know if it works for you.

Amazing that you are able to use particle swarms with the power flow! Let me know about your results with this. Usually I am concerned about the combinatorics taking too long.

miek770, 2019-08-05:
Great thanks! It works quite well on small grids at the moment, but bigger grids are very much a WiP. I'll keep you informed.

closed 2019-08-05 by miek770