SanPen/VeraGrid #113: Problems with running HELM

Hofsmo, 2019-10-01:
Hi,
I have recently been testing out GridCal. So far most things have been running very smoothly, except HELM.

I have tested with several different grids I have saved in the PSSE raw format as well as the five bus example provided in the GridCal documentation. In the case of the five bus example I merely changed the SolverType to HELM. In all cases I get the error below. Any idea what I could be doing wrong or if this is a known issue? In case I should provide more information please tell me.

```python
GridCal\Engine\Simulations\PowerFlow\helm_power_flow.py in get_rhs(n, V, W, Q, Vbus, Vst, Sbus, Pbus, nsys, nbus2, pv, pq, pvpos)
    147
    148     # compute the formulas
--> 149     f2 = Pbus[pv] * W[np.ix_(n-1, pv)] + QW_convolution
    150
    151     epsilon = -0.5 * VV_convolution

\numpy\lib\index_tricks.py in ix_(*args)
     97         new = asarray(new)
     98         if new.ndim != 1:
---> 99             raise ValueError("Cross index must be 1 dimensional")
    100         if new.size == 0:
    101             # Explicitly type empty arrays to avoid float default

ValueError: Cross index must be 1 dimensional
```

SanPen, 2019-10-01:
In general the HELM method is in alpha stage. Another person @Bengt is heaviliy working with HELM.
He made a pull request that I had to reject (is there still on hold), but I believe he's making progress with it.

Nonetheless I'll investigate the bug.

Hofsmo, 2019-10-01:
@SanPen Many thanks for the reply. I tried to install Bengt's master branch from source, but none of my code runs using his master branch. My code runs on your master branch, but with the same error as before. I'll check out some of Bengt's examples to see if I can make them run and maybe understand the error,

Bengt, 2019-10-01:
Hi guys,

I pretty much stopped working on HELM again. I got everything working that
I wanted, so I consider it done. The residual errors are beyond my
competence, because one would need to modify the actual algorithm. I am
only a user of grid calculation and thus have no interest in cracking into
that. So if you want to continue where I left of, feel free to do so. I
will not create further complications by changing my form for a while at
least.

Regards,
Bengt

2019, 14:59:

SanPen, 2019-10-03:
Hi, 

I fixed the issue, however I agree with @Bengt  that the method itself has very many pitfalls and better algorithms should be developed.

BR,
Santiago

Hofsmo, 2019-10-04:
Many thanks :) 

Btw. do you think that the problem is with the particular HELM implementation used in GridCal, or that it could be a more general problem related to HELM?

Cheers Sigurd

SanPen, 2019-10-04:
I guess that you have read the wild claims in the papers...

Up to 2017, I did program every single published HELM implementation and none worked like advertised, except maybe the vanilla (PQ nodes only) version. I even had to correct some of them to be half correct since they published formulation was plain wrong.

In general people are very secretive with this. In my oppinion the problem with HELM is the numeric precission required. The algorithm demands a very high numeric precission (double floats are not enough) and I believe that a path of research is to reduce the numeric precission demand by using tricks like the padé approximation.

By publishing the original code I was expecting htat someone would take the challenge and report improved working code back, but I have had no very little success.

Bengt, 2019-10-04:
Hi,

I have the same view @SanPen describes, here. I was interested in the properties of the algorithms, but they have many pitfalls that should to be addressed. I do not have the expertise required to work on the algorithms themselves, but I could contribute to improving their test coverage. I also noticed that while the theoretical complexity of HELM is generally favorable over Newton-Raphson-style methods, the empirical  runtime is way higher even for quite large grids (like 3000 nodes). So the applicability of HELM is limited to very large grids or if the NR methods cause you trouble. That does not mean however, that effort spend on HELM is futile. I think there is a very real and increasing demand for methods being able to solve power flow study for large grids quickly and do so reliably.

Regards,
Bengt

Hofsmo, 2019-10-07:
Hi,
Many thanks for your insights on the HELM methods. I am interested in the methods due to their claimed superior convergence properties. However, since none of the authors of the original papers have published their code it's difficult to know. I was therefore happy to see that you had implemented HELM. However, I had some doubts that it would really work as well as the authors claim when trying myself. So it was interesting to see that you guys confirm this.

Btw. I remember two papers talking these numerical problems and having claimed to solve them:
H. Chiang, T. Wang, and H. Sheng, “A novel fast and flexible holomorphic embedding power flow
method,” IEEE Transactions on Power Systems, vol. 33, no. 3, pp. 2551–2562, May 2018, ISSN: 0885-
8950. DOI: 10.1109/TPWRS.2017.2750711.

F. D. Freitas, A. C. Santos, L. F. J. Fernandes, and Y. G. I. Acle, “Restarted holomorphic embedding
load-flow model based on low-order padé approximant and estimated bus power injection,” International
Journal of Electrical Power & Energy Systems, vol. 112, pp. 326–338, Nov. 1, 2019, ISSN: 0142-0615.
DOI: 10.1016/j.ijepes.2019.04.051.

Did you also test these versions? As always I remain sceptical until someone else than the authors can provide the same results. At the moment I don't think I will have time to start implementing HELM methids, but I will let you guys know if I get time to give it a try.

SanPen, 2019-10-07:
Hi Sigurd, 

I have not tested those articles. To be honest I lost the faith in the new helm articles beign anything other than fancy lies. But I will read those and I'll see if there is redemption there :D

Since the Helm days, I have been putting more interest in robust iterative methods like better newton-raphson and Levenberg marquardt with quite positive results. Another method that I like very much is the Linear AC one, because it works always, and the error is comparable to the DC one.

Working in the NR algorithm, this weekend I managed to improve the performance a lot by using a smart line search trick. You can see the source by yourself.

BR,
Santiago

Hofsmo, 2019-10-07:
Interesting I will have a look at it. I just wrote a short memo on power flow convergence in a project, so I am very psyched and grateful that the methods are being implemented in a tool I have access to.

Btw. regarding the Levenberg-Marquardt I have seen people report its performance to be better than NR, but I have also seen some criticism. 
F. Milano, “Analogy and convergence of levenberg’s and lyapunov-based methods for power flow analysis,” IEEE Transactions on Power Systems, vol. 31, no. 2, pp. 1663–1664, Mar. 2016, ISSN: 0885-8950. DOI: 10.1109/TPWRS.2015.2415455.

Myself I want to try to implement iterative methods based on numerical integration. The idea is from the following article:
F. Milano, “Continuous newton’s method for power flow analysis,” IEEE Transactions on Power Systems, vol. 24, no. 1, pp. 50–57, Feb. 2009, ISSN: 0885-8950. DOI: 10.1109/TPWRS.2008.2004820.
I have other references following this line of research if you are interested in reading them. I think I will have an easier job convincing my boss to implement something following this line than HELM.

BR,
Sigurd

SanPen, 2019-10-07:
Indeed the Milano aproach was the most rfreshing thing I've read on the topic. I remember doing a dummy implementation and I did not manage to make it better since from what I remember there were several Jacobian calls per iteration.

Regarding the LM algorithm I have found that it is not the ultimate solution but rather a method that may work where NR fails. Just that.

In GridCal I programmed the following sequence when the methods fail to converge:
NR, Iwamoto, LM, Linear AC. This way, I know I'll have some solution if it exists. The aeeror achieved is another matter.

If you want to do research on new methods, I am usually interested in trying and be of any help I can.

BR,
Santiago

Hofsmo, 2019-10-07:
Nice,
Thanks for the insight. One of my projects are about to start and looking into new methods is one thing I am interested in. I hope I can convince my colleagues to pursue this. In which case I think GridCal can be a good starting point since you already have implemented several methods. I think it's better to have the different methods in the same tool if one wants to compare them. It is also better in the sense that one has a better possibility to find a solution if it exists.

BR,
Sigurd

SanPen, 2019-10-07:
Well just for the sake of the example, I have done this plot. The script producing it is [this](https://github.com/SanPen/GridCal/blob/master/src/research/power_flow/newton_line_search.py)
I just found where Milano's version of the algorithm was and I included it on the file as runge_kutta_nr.

[image]

Hofsmo, 2019-10-07:
Nice, that was fast. 
This is good inspiration to look further into these things.

Btw. as you already may expect my code runs with the different HELM implementations now. I will give the RK-method a try too.

BR,
Sigurtd

closed 2019-11-27 by SanPen

SanPen, 2020-02-25:
Hi guys, 

I have implemented a new HELM version that works way better.
There are still some weirdness with the reactive power but the series are perfectly convergent at least.

br,
Santiago

Bengt, 2020-02-25:
Hi, @SanPen!

You must be talking about Josep Fanals Batllori and this file:

https://github.com/SanPen/GridCal/blob/c2b2fa473b5fb01f8d18ac9cf78cf98c9a20fdc7/src/research/power_flow/helm/helm_josep.py

Thanks for the improved version. I would like to give it a spin later. I already had a look at the code, though, This implementation is also very convoluted and would benefit from a refactoring into smaller functions. Refactoring it is hindered by the code being untested as far as I can see. Also this makes it hard to maintain for you and thus hard to rely upon for users. Could your graduate student perhaps also provide some examples of the intended usage in the form of tests?

SanPen, 2020-02-25:
Hi, 

You're welcome to come up with a better implementation, but I believe the current one follows the formulation and it is not longer than say the line search Newton-Raphson.

So far, Josep and I are testing and doing the research here: https://github.com/u1946589/Holomorphic-Embedding-Josep

For the record, Josep is not my graduate student (nor Am I in academia whatsoever), he just submitted the draft code and we worked out this formulation, which by the way beats Newton-Raphson even for very large grids. Our focus at the moment is to investigate the effect of the approximations Sigma Vs. Padè Vs. Wynn and to come up with a convergence tractability test.

Best regards,
Santiago