SanPen/VeraGrid #117: Unable to run GridCal due to KeyError

Sibiriak, 2020-01-25:
Created empty Anaconda environment, installed GridCal via "pip install GridCal", activated the environment, then I run file with a code

`pip install GridCal`

 and got this:

> KLU failed
> Pardiso failed
> UmfPack failed
> Falling back to Blas/Lapack
> Using Blas/Lapack
> Traceback (most recent call last):
>   File "F:\Visual Studio 2017\gridcal\gridcal_test\gridcal_test.py", line 1, in <module>
>     from GridCal.ExecuteGridCal import run
>   File "C:\ProgramData\Anaconda3\envs\gridcal\lib\site-packages\GridCal\ExecuteGridCal.py", line 23, in <module>
>     from GridCal.Gui.Main.GridCalMain import run
>   File "C:\ProgramData\Anaconda3\envs\gridcal\lib\site-packages\GridCal\Gui\Main\GridCalMain.py", line 20, in <module>
>     from GridCal.Gui.GridEditorWidget import *
>   File "C:\ProgramData\Anaconda3\envs\gridcal\lib\site-packages\GridCal\Gui\GridEditorWidget\__init__.py", line 15, in <module>
>     from GridCal.Gui.GridEditorWidget.editor import *
>   File "C:\ProgramData\Anaconda3\envs\gridcal\lib\site-packages\GridCal\Gui\GridEditorWidget\editor.py", line 23, in <module>
>     from GridCal.Engine.Core.multi_circuit import MultiCircuit
>   File "C:\ProgramData\Anaconda3\envs\gridcal\lib\site-packages\GridCal\Engine\__init__.py", line 23, in <module>
>     from GridCal.Engine.IO import *
>   File "C:\ProgramData\Anaconda3\envs\gridcal\lib\site-packages\GridCal\Engine\IO\__init__.py", line 13, in <module>
>     from GridCal.Engine.IO.excel_interface import *
>   File "C:\ProgramData\Anaconda3\envs\gridcal\lib\site-packages\GridCal\Engine\IO\excel_interface.py", line 37, in <module>
>     def get_objects_dictionary(circuit=MultiCircuit()):
>   File "C:\ProgramData\Anaconda3\envs\gridcal\lib\site-packages\GridCal\Engine\Core\multi_circuit.py", line 86, in __init__
>     self.user_name = get_system_user()
>   File "C:\ProgramData\Anaconda3\envs\gridcal\lib\site-packages\GridCal\Engine\Core\multi_circuit.py", line 38, in get_system_user
>     user = getUser()
>   File "C:\ProgramData\Anaconda3\envs\gridcal\lib\site-packages\GridCal\Engine\Core\multi_circuit.py", line 36, in <lambda>
>     getUser = lambda: os.environ["USERNAME"] if "C:" in os.getcwd() else os.environ["USER"]
>   File "C:\ProgramData\Anaconda3\envs\gridcal\lib\os.py", line 679, in __getitem__
>     raise KeyError(key) from None
> KeyError: 'USER'

What do I do wrong?

SanPen, 2020-01-25:
Probably nothing. This looks like a bug, the like retrieves the username under windows... What windows version are you using?

SanPen, 2020-01-25:
A quick fix for you is to replace the function `get_system_user` in the file `multi_circuit.py` by this one:

```
def get_system_user():
    """
    Get the system mac + user name
    :return: string with the system mac address and the current user
    """

    # get the proper function to find the user depending on the platform
    if 'USERNAME' in os.environ:
        user = os.environ["USERNAME"]
    elif 'USER' in os.environ:
        user = os.environ["USER"]
    else:
        user = ''

    try:
        mac = get_mac()
    except:
        mac = ''

    return str(mac) + ':' + user
```

SanPen, 2020-01-25:
The problem arises because there is no 'USERNAME' in the path. This is probably caused by using environments. I don't know why people love environments since they are nothing but trouble.
Tell me if you succeed fixing this in order to close the issue.

BR,
Santiago

Sibiriak, 2020-01-25:
> Probably nothing. This looks like a bug, the line retrieves the username under windows... What windows version are you using?

Windows 10 X64.
Your suggested replacement for "get_system_user" worked for me, now it's fine. Thank you!

SanPen, 2020-01-25:
Excellent!
The change is committed and will be in the next release. 
Thank you!

adjaa, 2020-10-22:
Hello!
I have some trouble with gridCal installation. 
I ran "pip3 install GridCal" on kali 2020.3 and, I also tried to install it manually with the downloaded file (GridCal-3.7.0.tar.gz).
But always have the same error! Any idea? 

```
**run()"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/kali/.local/lib/python3.8/site-packages/GridCal/ExecuteGridCal.py", line 23, in <module>
    from GridCal.Gui.Main.GridCalMain import run
  File "/home/kali/.local/lib/python3.8/site-packages/GridCal/Gui/Main/GridCalMain.py", line 19, in <module>
    from GridCal.Gui.Main.MainWindow import *
  File "/home/kali/.local/lib/python3.8/site-packages/GridCal/Gui/Main/MainWindow.py", line 11, in <module>
    from PySide2 import QtCore, QtGui, QtWidgets
ImportError: /usr/local/lib/python3.8/dist-packages/PySide2/Qt/lib/libQt5Network.so.5: undefined symbol: _ZdaPvm, version Qt_5**
```
Thanks in advance!

SanPen, 2020-10-22:
Funky stuff.

I have just downloaded a virtualbox image of Kali linux, and it does come with a lot of python3 packages, among them PyQt5. This screws things up quite a bit.

In the same spirit, I downloaded the [standalone version of GridCal](https://gridcal.readthedocs.io/en/latest/getting_started/install.html) and installed 
the missing library libffi6:

```
wget http://mirrors.kernel.org/ubuntu/pool/main/libf/libffi/libffi6_3.2.1-8_amd64.deb
sudo apt install ./libffi6_3.2.1-8_amd64.deb
```
And then run GridCal with the script provided: `RunGridCal.sh`, et voilà it works.

Tell me if you manage.

adjaa, 2020-10-22:
Thanks a lot, it works!
RunGridCal.sh shows the gui.
But the module `(from GridCal.Engine import * )` is not found when I import it and, I want to use GridCal as a library!

SanPen, 2020-10-23:
Hi, 

You need to run the standalone python, not the system python.
I have just tried on my Kali VM and it works.

I recommend that you use an IDE such as PyCharm and there select the python interpreter to use.
That is what I do.

adjaa, 2020-10-23:
Hi, 
that's okay!
thanks for your help!

adjaa, 2020-10-26:
Hello SanPen,
sorry to bother you again!
I run some GridCal script with PyCharm and have this error:
```
/home/kali/Downloads/GridCal_standalone/bin/python3 /home/kali/PycharmProjects/testGrid/main.py
KLU failed
Pardiso failed
UmfPack failed
Falling back to Blas/Lapack
Using Blas/Lapack

Process finished with exit code 0
```
I used the same configuration of this issue _(https://github.com/SanPen/GridCal/issues/118)_ and also tried to install PyQt5. 
```
kali@kali:~/PycharmProjects/testGrid$ pip install PyQt5
Requirement already satisfied: PyQt5 in /usr/lib/python3/dist-packages (5.15.0)
```
The package is already installed but not in the GridCal interpreter and I don't know how to copy it! 
Any idea?

SanPen, 2020-10-26:
Hi!,

Not sure what the problem is, when a program exist with `0` it is because all went ok.

Maybe you can share what you are trying, but check that you're actually  printing the output or whatever it that you need.

About PyQt5, you should never install PyQt5 because of GridCal. It will always be the PySide2 package the one in charge of the Qt stuff.

BR,
Santiago

adjaa, 2020-10-26:
You're right, actually I have the output of the calculation. 
I choose one of your test script and run it to have the grid shown as a "plot".

```
import numpy as np
from GridCal.Engine.Core.multi_circuit import MultiCircuit
from GridCal.Engine.Devices.branch import Branch
from GridCal.Engine.Devices.bus import Bus
from GridCal.Engine.Devices.generator import Generator
from GridCal.Engine.Devices.load import Load
from GridCal.Engine.Simulations.PowerFlow.power_flow_worker import SolverType
from GridCal.Engine.Simulations.PowerFlow.power_flow_driver import PowerFlowOptions, PowerFlowDriver
from GridCal.Gui.GuiFunctions import ResultsModel

from print_power_flow_results import print_power_flow_results
from conftest import ROOT_PATH

def test_demo_5_node(root_path=ROOT_PATH):
    np.core.arrayprint.set_printoptions(precision=4)

    grid = MultiCircuit()

    # Add buses
    bus1 = Bus('Bus 1', vnom=20)
    grid.add_bus(bus1)
    gen1 = Generator('Slack Generator', voltage_module=1.0)
    grid.add_generator(bus1, gen1)

    bus2 = Bus('Bus 2', vnom=20)
    grid.add_bus(bus2)
    grid.add_load(bus2, Load('load 2', P=40, Q=20))

    bus3 = Bus('Bus 3', vnom=20)
    grid.add_bus(bus3)
    grid.add_load(bus3, Load('load 3', P=25, Q=15))

    bus4 = Bus('Bus 4', vnom=20)
    grid.add_bus(bus4)
    grid.add_load(bus4, Load('load 4', P=40, Q=20))

    bus5 = Bus('Bus 5', vnom=20)
    grid.add_bus(bus5)
    grid.add_load(bus5, Load('load 5', P=50, Q=20))

    # add branches (Lines in this case)
    grid.add_branch(Branch(bus1, bus2, 'line 1-2', r=0.05, x=0.11, b=0.02))
    grid.add_branch(Branch(bus1, bus3, 'line 1-3', r=0.05, x=0.11, b=0.02))
    grid.add_branch(Branch(bus1, bus5, 'line 1-5', r=0.03, x=0.08, b=0.02))
    grid.add_branch(Branch(bus2, bus3, 'line 2-3', r=0.04, x=0.09, b=0.02))
    grid.add_branch(Branch(bus2, bus5, 'line 2-5', r=0.04, x=0.09, b=0.02))
    grid.add_branch(Branch(bus3, bus4, 'line 3-4', r=0.06, x=0.13, b=0.03))
    grid.add_branch(Branch(bus4, bus5, 'line 4-5', r=0.04, x=0.09, b=0.02))
    #grid.plot_graph()
    print('\n\n', grid.name)

    options = PowerFlowOptions(SolverType.NR, verbose=False)

    power_flow = PowerFlowDriver(grid, options)
    power_flow.run()

    print_power_flow_results(power_flow=power_flow)
    v = np.array([1., 0.9553, 0.9548, 0.9334, 0.9534])
    all_ok = np.isclose(np.abs(power_flow.results.voltage), v, atol=1e-3)
    mdl = power_flow.results.mdl()
    mdl.plot()
    return all_ok

if __name__ == '__main__':
    test_demo_5_node(root_path=ROOT_PATH)
```
I have the values but can't print the corresponding graph. May be I do it in wrong way?

SanPen, 2020-10-27:
Hi, 

You need to import matplotlib:

`from matplotlib import pyplot as plt`

And at the end of all, write:

`plt.show()`

adjaa, 2020-10-27:
I add it:
```
import numpy as np
from GridCal.Engine.Core.multi_circuit import MultiCircuit
from GridCal.Engine.Devices.branch import Branch
from GridCal.Engine.Devices.bus import Bus
from GridCal.Engine.Devices.generator import Generator
from GridCal.Engine.Devices.load import Load
from GridCal.Engine.Simulations.PowerFlow.power_flow_worker import SolverType
from GridCal.Engine.Simulations.PowerFlow.power_flow_driver import PowerFlowOptions, PowerFlowDriver
from GridCal.Gui.GuiFunctions import ResultsModel

from print_power_flow_results import print_power_flow_results
from conftest import ROOT_PATH
from matplotlib import pyplot as plt

def test_demo_5_node(root_path=ROOT_PATH):
    np.core.arrayprint.set_printoptions(precision=4)

    grid = MultiCircuit()

    # Add buses
    bus1 = Bus('Bus 1', vnom=20)
    grid.add_bus(bus1)
    gen1 = Generator('Slack Generator', voltage_module=1.0)
    grid.add_generator(bus1, gen1)

    bus2 = Bus('Bus 2', vnom=20)
    grid.add_bus(bus2)
    grid.add_load(bus2, Load('load 2', P=40, Q=20))

    bus3 = Bus('Bus 3', vnom=20)
    grid.add_bus(bus3)
    grid.add_load(bus3, Load('load 3', P=25, Q=15))

    bus4 = Bus('Bus 4', vnom=20)
    grid.add_bus(bus4)
    grid.add_load(bus4, Load('load 4', P=40, Q=20))

    bus5 = Bus('Bus 5', vnom=20)
    grid.add_bus(bus5)
    grid.add_load(bus5, Load('load 5', P=50, Q=20))

    # add branches (Lines in this case)
    grid.add_branch(Branch(bus1, bus2, 'line 1-2', r=0.05, x=0.11, b=0.02))
    grid.add_branch(Branch(bus1, bus3, 'line 1-3', r=0.05, x=0.11, b=0.02))
    grid.add_branch(Branch(bus1, bus5, 'line 1-5', r=0.03, x=0.08, b=0.02))
    grid.add_branch(Branch(bus2, bus3, 'line 2-3', r=0.04, x=0.09, b=0.02))
    grid.add_branch(Branch(bus2, bus5, 'line 2-5', r=0.04, x=0.09, b=0.02))
    grid.add_branch(Branch(bus3, bus4, 'line 3-4', r=0.06, x=0.13, b=0.03))
    grid.add_branch(Branch(bus4, bus5, 'line 4-5', r=0.04, x=0.09, b=0.02))
    #grid.plot_graph()
    print('\n\n', grid.name)

    options = PowerFlowOptions(SolverType.NR, verbose=False)

    power_flow = PowerFlowDriver(grid, options)
    power_flow.run()

    print_power_flow_results(power_flow=power_flow)
    v = np.array([1., 0.9553, 0.9548, 0.9334, 0.9534])
    all_ok = np.isclose(np.abs(power_flow.results.voltage), v, atol=1e-3)
    mdl = power_flow.results.mdl()
    mdl.plot()
    plt.show()
    return all_ok

if __name__ == '__main__':
    test_demo_5_node(root_path=ROOT_PATH)
    
```

but have this error:
```
Traceback (most recent call last):
  File "/home/kali/PycharmProjects/testGrid/main.py", line 70, in <module>
    test_demo_5_node(root_path=ROOT_PATH)
  File "/home/kali/PycharmProjects/testGrid/main.py", line 62, in test_demo_5_node
    mdl = power_flow.results.mdl()
TypeError: mdl() missing 1 required positional argument: 'result_type'

Process finished with exit code 1
```

SanPen, 2020-10-27:
Hi, 

So as the error says, you're missing what kind of result yu want the model of.
So:

`mdl = power_flow.results.mdl(ResultTypes.BusVoltageModule)`

Maybe?

Check the available results with 

`power_flow.results.available_results`

In case results types are not imported:

`from GridCal.Engine.Simulations.result_types import ResultTypes`

adjaa, 2020-10-27:
That's okay,
Thanks for your help!

SanPen, 2020-10-28:
you're welcome :D

closed 2020-10-28 by SanPen