SanPen/VeraGrid #118: Unable to run GridCal after installation

anmold96, 2020-02-23:
Hello, 

I have successfully installed GridCal by installing anaconda, however, I'm unable to run it on my windows 10. Whenever I run it I get the following error: 
---------------------------------------------------------------------------------------------------------
KLU failed
Pardiso failed
UmfPack failed
Falling back to Blas/Lapack
Using Blas/Lapack
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\GridCal\ExecuteGridCal.py", line 23, in <module>
    from GridCal.Gui.Main.GridCalMain import run
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\GridCal\Gui\Main\GridCalMain.py", line 20, in <module>
    from GridCal.Gui.GridEditorWidget import *
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\GridCal\Gui\GridEditorWidget\__init__.py", line 15, in <module>
    from GridCal.Gui.GridEditorWidget.editor import *
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\GridCal\Gui\GridEditorWidget\editor.py", line 23, in <module>
    from GridCal.Engine.Core.multi_circuit import MultiCircuit
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\GridCal\Engine\__init__.py", line 19, in <module>
    from GridCal.Engine.grid_analysis import *
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\GridCal\Engine\grid_analysis.py", line 18, in <module>
    from GridCal.Engine.Simulations.PowerFlow.time_series_driver import TimeSeriesResults
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\GridCal\Engine\Simulations\__init__.py", line 18, in <module>
    from GridCal.Engine.Simulations.Stochastic import *
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\GridCal\Engine\Simulations\Stochastic\__init__.py", line 16, in <module>
    from GridCal.Engine.Simulations.Stochastic.lhs_driver import *
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\GridCal\Engine\Simulations\Stochastic\lhs_driver.py", line 21, in <module>
    from GridCal.Engine.Simulations.PowerFlow.power_flow_results import PowerFlowResults
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\GridCal\Engine\Simulations\PowerFlow\__init__.py", line 2, in <module>
    from GridCal.Engine.Simulations.PowerFlow.power_flow_worker import *
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\GridCal\Engine\Simulations\PowerFlow\power_flow_worker.py", line 22, in <module>
    from GridCal.Engine.Simulations.PowerFlow.jacobian_based_power_flow import IwamotoNR
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\GridCal\Engine\Simulations\PowerFlow\jacobian_based_power_flow.py", line 27, in <module>
    from GridCal.Engine.Simulations.PowerFlow.numba_functions import calc_power_csr_numba, diag
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\GridCal\Engine\Simulations\PowerFlow\numba_functions.py", line 23, in <module>
    def calc_power_csr_numba(n, Yp, Yj, Yx, V, I, n_par=500):
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\numba\decorators.py", line 200, in wrapper
    disp.compile(sig)
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\numba\compiler_lock.py", line 32, in _acquire_compile_lock
    return func(*args, **kwargs)
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\numba\dispatcher.py", line 768, in compile
    cres = self._compiler.compile(args, return_type)
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\numba\dispatcher.py", line 77, in compile
    status, retval = self._compile_cached(args, return_type)
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\numba\dispatcher.py", line 91, in _compile_cached
    retval = self._compile_core(args, return_type)
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\numba\dispatcher.py", line 109, in _compile_core
    pipeline_class=self.pipeline_class)
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\numba\compiler.py", line 551, in compile_extra
    return pipeline.compile_extra(func)
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\numba\compiler.py", line 331, in compile_extra
    return self._compile_bytecode()
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\numba\compiler.py", line 393, in _compile_bytecode
    return self._compile_core()
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\numba\compiler.py", line 373, in _compile_core
    raise e
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\numba\compiler.py", line 364, in _compile_core
    pm.run(self.state)
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\numba\compiler_machinery.py", line 347, in run
    raise patched_exception
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\numba\compiler_machinery.py", line 338, in run
    self._runPass(idx, pass_inst, state)
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\numba\compiler_lock.py", line 32, in _acquire_compile_lock
    return func(*args, **kwargs)
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\numba\compiler_machinery.py", line 302, in _runPass
    mutated |= check(pss.run_pass, internal_state)
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\numba\compiler_machinery.py", line 275, in check
    mangled = func(compiler_state)
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\numba\typed_passes.py", line 407, in run_pass
    NativeLowering().run_pass(state)
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\numba\typed_passes.py", line 349, in run_pass
    lower.lower()
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\numba\lowering.py", line 195, in lower
    self.lower_normal_function(self.fndesc)
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\numba\lowering.py", line 248, in lower_normal_function
    entry_block_tail = self.lower_function_body()
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\numba\lowering.py", line 273, in lower_function_body
    self.lower_block(block)
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\numba\lowering.py", line 288, in lower_block
    self.lower_inst(inst)
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\numba\lowering.py", line 476, in lower_inst
    func(self, inst)
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\numba\npyufunc\parfor.py", line 52, in _lower_parfor_parallel
    ensure_parallel_support()
  File "C:\Users\Anmol\Anaconda3\lib\site-packages\numba\parfor.py", line 4156, in ensure_parallel_support
    raise errors.UnsupportedParforsError(msg)
numba.errors.UnsupportedParforsError: Failed in nopython mode pipeline (step: nopython mode backend)
[1m[1mThe 'parallel' target is not currently supported on Windows operating systems when using Python 2.7, or on 32 bit hardware.[0m
[0m[1m[1] During: lowering "id=0[LoopNest(index_variable = parfor_index.15, range = (0, n, 1))]{277: <ir.Block at C:\Users\Anmol\Anaconda3\lib\site-packages\GridCal\Engine\Simulations\PowerFlow\numba_functions.py (38)>}Var(parfor_index.15, numba_functions.py:38)" at C:\Users\Anmol\Anaconda3\lib\site-packages\GridCal\Engine\Simulations\PowerFlow\numba_functions.py (38)[0m
-------------------------------------------------------------------------------------------------------------

Could you please help me here? I'm not sure what I'm doing wrong.

SanPen, 2020-02-23:
Hi!
I see some issues there:
- I see python 2.7, 32 bits....it must be 3.6 or newer for 64 bits
- Anaconda is cool for some rhings but they have do e a really poor job supporting Qt, tvhe graphic library...

So if you want to use GridCal as a library, I recomend removing Anaconda and installing python 3.7. Then install GridCal and all the livraries will be fine.

Otherwise just download the standalone GridCal that I provide.

Read the installation part on the documentation if you need more details.

best regards,
Santiago

anmold96, 2020-02-23:
Hello, Thank for your response. 

I'm unable to download the standalone setup file for windows. When I click the on the link provided it shows the following: 

`https://sanpv.files.wordpress.com/2019/05/gridcalsetup-1.zip might be temporarily down or it may have moved permanently to a new web address.`

anmold96, 2020-02-24:
Also, how do I deal with this error?
`ModuleNotFoundError: No module named 'numba.pycc'`

I have already installed the numba package but GridCal just stops here every time..

SanPen, 2020-02-24:
About Numba, I have no clue, looks like a compiler thing...
I am serving the setup with this link since wordpress is having issues with the installer:
https://drive.google.com/file/d/1_ln73z0pKYcxw8TEYroWXTqK6ZYLY6hS/view?usp=drivesdk

please try

SanPen, 2020-02-25:
Did you succeed?

anmold96, 2020-02-26:
Hello, I got the pip version up and running. 
Here's what I actually did: I uninstalled the 32 bit anaconda and installed the 64 bit version. Then, while running GridCal using pip, I was getting the THE NO MOULE FOUND NUMBA.PYCC error. I reinstalled the numba module and a couple of other libraries (upgraded Qt) and then the GUI worked perfectly. I didn't try out the link you posted because it is working just fine with pip as of now. In the future if it stops running for some reason, then I'll use the standalone version you posted. Thanks much for the help.

SanPen, 2020-02-27:
I am glad it worked.
Also, I appreciate all the feedback that you may provide.
Best regards, 
Santiago

anmold96, 2020-03-01:
> About Numba, I have no clue, looks like a compiler thing...
> I am serving the setup with this link since wordpress is having issues with the installer:
> https://drive.google.com/file/d/1_ln73z0pKYcxw8TEYroWXTqK6ZYLY6hS/view?usp=drivesdk
> 
> please try

Hello, since I wanted to develop my own scripts using GridCal, I installed the standalone version. I wasn't able to use both spyder and GridCal to develop my own scripts due to an error already discussed in the issues section a few months ago. The GUI is working fine after the standalone download but I'm not sure how and where should I write my scripts. Could you please guide me here?

SanPen, 2020-03-01:
Hi,

The way I develop is with an IDE called PyCharm. With it you can chose the python executable that you want go use, hence you would be able to choose the standalone python from GridCal. 
It takes a while to get used ro, but it is the best way to develop pythom IMHO.

anmold96, 2020-03-01:
> Hi,
> 
> The way I develop is with an IDE called PyCharm. With it you can chose the python executable that you want go use, hence you would be able to choose the standalone python from GridCal.
> It takes a while to get used ro, but it is the best way to develop pythom IMHO.

This is the kind of error I am getting. Although I'm able to run the scripts (GridCal as a calculator is working), I'm not able to see any visualizations or graphs. What can be done now? I have set the python interpreter to standalone python.

Backend Qt5Agg is interactive backend. Turning interactive mode on.

KLU failed
Pardiso failed
UmfPack failed
Falling back to Blas/Lapack
Using Blas/Lapack

Failed to enable GUI event loop integration for 'qt5'
Traceback (most recent call last):
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2019.3.3\plugins\python-ce\helpers\pydev\_pydev_bundle\pydev_code_executor.py", line 211, in do_enable_gui
    enable_gui(guiname)
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2019.3.3\plugins\python-ce\helpers\pydev\pydev_ipython\inputhook.py", line 536, in enable_gui
    return gui_hook(app)
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2019.3.3\plugins\python-ce\helpers\pydev\pydev_ipython\inputhook.py", line 221, in enable_qt5
    from pydev_ipython.inputhookqt5 import create_inputhook_qt5
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2019.3.3\plugins\python-ce\helpers\pydev\_pydev_bundle\pydev_import_hook.py", line 21, in do_import
    module = self._system_import(name, *args, **kwargs)
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2019.3.3\plugins\python-ce\helpers\pydev\pydev_ipython\inputhookqt5.py", line 25, in <module>
    from PyQt5 import QtCore, QtGui
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2019.3.3\plugins\python-ce\helpers\pydev\_pydev_bundle\pydev_import_hook.py", line 21, in do_import
    module = self._system_import(name, *args, **kwargs)

ModuleNotFoundError: No module named 'PyQt5'
Failed to enable GUI event loop integration for 'qt5'
Traceback (most recent call last):
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2019.3.3\plugins\python-ce\helpers\pydev\_pydev_bundle\pydev_code_executor.py", line 211, in do_enable_gui
    enable_gui(guiname)
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2019.3.3\plugins\python-ce\helpers\pydev\pydev_ipython\inputhook.py", line 536, in enable_gui
    return gui_hook(app)
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2019.3.3\plugins\python-ce\helpers\pydev\pydev_ipython\inputhook.py", line 221, in enable_qt5
    from pydev_ipython.inputhookqt5 import create_inputhook_qt5
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2019.3.3\plugins\python-ce\helpers\pydev\_pydev_bundle\pydev_import_hook.py", line 21, in do_import
    module = self._system_import(name, *args, **kwargs)
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2019.3.3\plugins\python-ce\helpers\pydev\pydev_ipython\inputhookqt5.py", line 25, in <module>
    from PyQt5 import QtCore, QtGui
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2019.3.3\plugins\python-ce\helpers\pydev\_pydev_bundle\pydev_import_hook.py", line 21, in do_import
    module = self._system_import(name, *args, **kwargs)
ModuleNotFoundError: No module named 'PyQt5' `

SanPen, 2020-03-01:
I am using this configuration:

[image]

So with this it may be fixed.

anmold96, 2020-03-01:
[image]
 Getting this error now

SanPen, 2020-03-01:
Hi, 

I see that you're trying to run scripts from the interactive python console. This is something I never do, since I write scripts and run them in normal operation mode.

However, if you install PyQt5, you will be able to use the ipython console. I do not ship PySide5 and PyQt5 with GridCal's python distro because in the normal operation this is not required, but you may always extend the distro with the packages you need of course.

You can manage the distribution packaged directly from pycharm in the settings.
See Project > Project interpreter, and there add packages with the + simbol

Best regards,
Santiago

anmold96, 2020-03-01:
If I run it using a python file then I get the following error: 

C:\GridCal\python\python.exe C:/Users/Anmol/.PyCharmCE2019.3/config/scratches/scratch.py
Traceback (most recent call last):
  File "C:/Users/Anmol/.PyCharmCE2019.3/config/scratches/scratch.py", line 1, in <module>
    from GridCal.Engine import *
  File "C:\GridCal\python\lib\site-packages\GridCal\Engine\__init__.py", line 18, in <module>
    from GridCal.Engine.Devices import *
  File "C:\GridCal\python\lib\site-packages\GridCal\Engine\Devices\__init__.py", line 17, in <module>
    from GridCal.Engine.Devices.branch import *
  File "C:\GridCal\python\lib\site-packages\GridCal\Engine\Devices\branch.py", line 29, in <module>
    from GridCal.Engine.Devices.tower import Tower
  File "C:\GridCal\python\lib\site-packages\GridCal\Engine\Devices\tower.py", line 19, in <module>
    from PySide2 import QtCore
ImportError: DLL load failed: The specified procedure could not be found.

Process finished with exit code 1

I didn't exactly get what were you recommending. Should I uninstall PyQt5?

SanPen, 2020-03-01:
nope, you should install it if you want to have iPython console. I have just done it and it worked for me on Windows10.

But my true recommendation is to drop the interactive console style of scripting and do it old-school by writing the script in a `.py` and running it.

anmold96, 2020-03-01:
It's working well now. Also, all the plots in the GUI like voltage magnitude, voltage angles, PV curves for continuation power flow; Can they also be plotted using scripting, for example, using pycharm without using the GUI ?

SanPen, 2020-03-01:
Absolutelly,

In order to plot anything in gridcal you perform the simulation with the corresponding driver.
In the case of voltage collapse:

```
vc_options = VoltageCollapseOptions()
    # just for this test
    numeric_circuit = main_circuit.compile_snapshot()
    numeric_inputs = numeric_circuit.compute()
    Sbase = np.zeros(len(main_circuit.buses), dtype=complex)
    Vbase = np.zeros(len(main_circuit.buses), dtype=complex)
    for c in numeric_inputs:
        Sbase[c.original_bus_idx] = c.Sbus
        Vbase[c.original_bus_idx] = c.Vbus
    unitary_vector = -1 + 2 * np.random.random(len(main_circuit.buses))
    vc_inputs = VoltageCollapseInput(Sbase=Sbase,
                                     Vbase=Vbase,
                                     Starget=Sbase * (1 + unitary_vector))
    vc = VoltageCollapse(circuit=main_circuit, options=vc_options,
                         inputs=vc_inputs)
    vc.run()
    mdl = vc.results.mdl()
    mdl.plot()
```

anmold96, 2020-03-06:
Say I want to plot the voltage magnitudes and angles using a script after running a power flow on a power system. How shall I do that?

SanPen, 2020-03-06:
Hi, 

You can try this:

```
options = PowerFlowOptions(SolverType.NR, verbose=False)
power_flow = PowerFlowDriver(grid, options)
power_flow.run()

vm = np.abs(power_flow.results.voltage)
va = np.angle(power_flow.results.voltage)
```

And then plot ´vm´ and ´va´.

closed 2020-04-20 by SanPen