SanPen/VeraGrid #119: NOSE versus FULL CURVE for voltage stability analysis

anmold96, 2020-03-01:
Hello, I'm currently using the GUI of GridCal for power systems analysis. After reading the documentation, running a power flow & CPF on a 5 bus system, I wanted to know the difference between the NOSE and the FULL CURVE setting for voltage stability. 

Ideally, I expect to get a half curve until the bifurcation point if I choose NOSE (similar to something blow)
[image]

& I expect to the entire curve if I choose FULL CURVE (similar to something below)
[image]

However, irrespective of choosing NOSE or FULL CURVE I only get half the curve similar to the first figure when I plot it using the GUI. Is my understanding correct regarding the settings?

SanPen, 2020-03-02:
You are correct. The functionality should be what you expect, however it is not. I cannot recall when did I modify it not to have the full curve. A couple of months ago I revisited the issue but I could not easily fix it.

Unfortunately I cannot promise a fix soon.

Maybe you can help me have a look at it

SanPen, 2020-03-06:
For your information, we are working in a very very new way of drawing the full curve using the HELM method. It will probably be ready for the next release.

anmold96, 2020-03-29:
vc_options = VoltageCollapseOptions()
numeric_circuit = grid.compile()
numeric_inputs = numeric_circuit.compute()
Sbase = np.zeros(len(grid.buses), dtype=complex)
Vbase = np.zeros(len(grid.buses), dtype=complex)
for c in numeric_inputs:
    Sbase[c.original_bus_idx] = c.Sbus
    Vbase[c.original_bus_idx] = c.Vbus
`unitary_vector = -1 + 2 * np.random.random(len(grid.buses))`
vc_inputs = VoltageCollapseInput(Sbase=Sbase,
                                 Vbase=Vbase,
                                 Starget=Sbase * (1 + unitary_vector))
vc = VoltageCollapse(circuit=grid, options=vc_options,
                     inputs=vc_inputs)

Hello!

In the code you shared for voltage collapse could you please tell me how are you deciding the `Starget`? To be specific, what is this statement is doing: `unitary_vector = -1 + 2 * np.random.random(len(grid.buses))` ?

I was trying to replicate the voltage stability results for an IEEE 5 bus system, So, on one hand, I'm running the GUI (by loading the file IEEE 5 Bus.xlsx on your GitHub as a circuit) and on the other hand, I'm scripting the same IEEE 5 Bus system in PyCHARM. 

The problem I'm facing is that every time I'm using the above code for voltage stability I'm getting different P-V curves for every run, however, in case of the GUI every time I run it I get the same P-V curves.  Could you explain what's going on with the scripting of voltage collapse?

SanPen, 2020-03-30:
Hi, 

The algorithm allows you to reach any bus loading combination that you may want. THe statement that you ask for defines a "target" state that is a linear combination of the initial state. This is done like this so that the algorithm increases the load equally among all the nodes which is not a strict requirement since you may very well want to assess a future situation that is not a linear combination.

Then the algorithm continues in the specified direction until collapse.

closed 2020-04-20 by SanPen