SanPen/VeraGrid #121: Creating a new grid profile in excel sheet

anmold96, 2020-03-16:
Hello, I had two specific questions: 

1. I wanted to know whether I could create a new customized power system data file by specifying the load, generation, and branches as input parameters using an excel sheet and then importing the model using the standard FileOpen(fname).open() function. 
I know that I can develop my custom circuit/grid using the GUI interface, but since I want to automate my scripts and hence, I wish to develop my own input parameter data file to further it in GridCal. 

2. I also wanted to know that for voltage collapse, does GridCal increase the load at all the buses simultaneously during the simulation or does it increase the load at each of the buses but one at a time? I mean is it like increase the load at one specific bus until voltage collapse, and then increase the load at another bus keeping the load at the previous bus to be constant, or does the load at all the buses increase simultaneously?
In the PST toolbox all the loads increase simultaneously by default, so wanted to confirm with you.

Thank you!

SanPen, 2020-03-16:
Hi!, 

1: Yes, although you wold have to do it like [this tutorial](https://github.com/SanPen/GridCal/blob/master/src/Tutorials/defining_a_grid_from_scratch_with_profiles.py)

2: Yes, all the loads vary together.

anmold96, 2020-04-11:
Hello, 

Thanks for the tutorial. 

I was building a power system model and wanted to connect a transformer with user-specified parameters. In the documentation, I came across this template of yours:

# CREATING A CUSTOM TRANSORMER MODEL
SS = TransformerType(name="SS",
                     hv_nominal_voltage=100, # kV
                     lv_nominal_voltage=10, # kV
                     nominal_power=100, # MVA
                     copper_losses=10000, # kW
                     iron_losses=125, # kW
                     no_load_current=0.5, # %
                     short_circuit_voltage=8) # %

# adding the template to the branch...
X_C3 = Branch(bus_from=POI,
              bus_to=B_C3,
              name="X_C3",
              rate=460,
              branch_type=BranchType.Transformer,
              template=SS,
              )

I wanted to confirm whether the "rate" in Branch Object and "nominal_power" in the TransformerType object have to be the same?

What's the difference between Branch.rate and TransformerType.nominal_power?

SanPen, 2020-04-11:
Hi, 

Yes, the transformer type nominal power is assigned to the branch rate.

For the transformer, the nominal power is whatever the transformer nameplate says (10 MVA, or 50 MVA, etc.) The transformer type is specified from the short circuit test of the transformer as you probably know. Then the per-unit branch values are derived using the system base (100 MVA usually), because the per-unit values must always be on the system base.

Check out the [transformer docs](https://gridcal.readthedocs.io/en/latest/theory/xfo_sc.html) for more information.

anmold96, 2020-04-11:
[image]

I have the following PSSE format for transformers

SanPen, 2020-04-11:
Hi, 

The PSS/e input format is particularly shitty because as you can see the values may either be on kV or p.u. From what you show, I can tell you that you are setting the branch reactance and then applying a template that has no reactance, hence "smashing" the specified one. 

In this case I would not use the template and only set the reactance `x=0.1 p.u.`

anmold96, 2020-04-11:
Actually when I'm setting the reactance to `x=0.1 p.u.` then the power flow results in GridCal does not match the power flow results with PSSE. What can be done in such a case?

SanPen, 2020-04-11:
As you may imagine I cannot asses whether the rest of the stuff is right...

You may export to .raw and import that in GridCal and check.

I can tell you for fact that IEEE14, 30 and 118 match PSS/e with up to 1e-6 precission.

anmold96, 2020-04-11:
You're right. For other files that I'm generating from PSSE are giving the same results as in GridCal. But for this particular bus system .raw file, I'm getting very different solutions. I'm don't want you to assess the rest, but all I'm doing is exporting from PSSE and importing in GridCal and checking and unfortunately getting different results.

SanPen, 2020-04-11:
Lately I have discovered the following issues in the PSS/e importer wich may be a cause of trouble:

- Repeated branches, meaning branches with the same bus_from, bus_to, ckt
- The buses marked as type 3 (slack) are not the same as the slack bus in the area
- The transformer in the raw file are not specified in p.u. but in the other ludicrous ways wich are unsupported by GridCal at the moment.

anmold96, 2020-04-11:
Okay, thanks for your time! I'll try to figure it out.

SanPen, 2020-04-11:
Good luck! sometimes it takes quite some time to see what is the issue

anmold96, 2020-04-11:
Could I know what is the function of `bus_to_regulated` exactly? 
I couldn't exactly figure it out by the code and documentation.

SanPen, 2020-04-11:
Hi, 

There is a feature where you tell the transformer to control the voltage at the "bus_to", if so and the power flow option to regulate transformers is enabled, then the transformer will try to control the voltage.

anmold96, 2020-04-13:
Okay, Thank you. 

I also wanted to know whether would it be possible to increase the load for continuation power flow analysis only for selected/specific load buses of my choice? Can this be done?

SanPen, 2020-04-13:
I would say that you can achieve that by setting the unitary vector to zeros except for the position of the buses that you want where you can put ones (or any value greater than 0)

anmold96, 2020-04-13:
It makes perfect sense. I just tried that out.  Thanks a lot.

SanPen, 2020-04-13:
excellent :)

closed 2020-04-20 by SanPen