SanPen/VeraGrid #122: Yseries and Yshunt Calculation

Zalnd, 2020-04-30:
Hi

The formulation implemented in 'snapshot_static_inputs.py' seems to have a problem because the following equation doesn't hold:

 Ybus = Yseries + Yshunt

My test system was Iwamoto11Bus.xlsx (https://forum.openmod-initiative.org/t/the-new-helm-powerflow-in-gridcal/1952) after changing the Bshunt in Bus 11 from 0.044 Mvar to 4.4 Mvar.

One obvious workaround is:
Yshunt = Ybus - Yseries
Yshunt = Yshunt.diagonal().reshape(-1)

Also, the test function in 'helm_power_flow.py' should make 'Ysh0' equal to 'inputs.Ysh_helm'.

SanPen, 2020-04-30:
Hi, 

On the contrary, you should convert Yshunt to a sparse matrix.

The easy way to do it is using the function `diags` from `scipy.sparse`

The it should be something like

`Ybus = Yseries + diags(Yshunt)`

Zalnd, 2020-04-30:
Well, I thought that 'calc_connectivity' had to return Yshunt as an one-dimensional array.

Anyway, the present independent calculation of Ybus, Yseries and Yshunt doesn't obey 'Ybus = Yseries + Yshunt' from the mathematical point of view. At least for the Iwamoto ill-conditioned system. You may check yourself that "np.max(np.abs(Ybus-Yseries-diags(Yshunt)))" gives the answer "00041723356008991144"

SanPen, 2020-05-01:
I honestly don't know what you are doing to come up with that since that particular calculation has been checked quite a few times in agreement with Josep's independent calculations.

Nonetheless, the common practice is to provide a minimal case where what you observe is happening so it may be sorted out.

But as always, you may check out the [code](https://github.com/SanPen/GridCal/blob/e789689a6b7674616b0ed2d795d54ba40d9255bd/src/GridCal/Engine/Core/series_static_inputs.py#L36):  function `calc_connectivity`

Zalnd, 2020-05-01:
[Iwamoto11Bus.xlsx](https://github.com/SanPen/GridCal/files/4564123/Iwamoto11Bus.xlsx)

I took a look at the mathematical formulation. It seems to be correct. Still, the equality between 'Ybus' and 'Yseries + Yshunt' doesn't hold.

This problem can be numerical since Iwamoto 11 Bus System is ill-conditioned. As I meant before, one obvious solution is to calculate two of them independently and then calculate the last one from the equality.

SanPen, 2020-05-06:
Check [this](https://github.com/SanPen/GridCal/blob/master/src/tests/admittance_matrix_test.py) out.

Chances are that you were looking at the wrong Yshunt.

Best regards,
Santiago

closed 2020-05-11 by SanPen