SanPen/VeraGrid #123: Unable to open PSSE-30 RAW file

rafikfahem, 2020-05-26:
Hi,

I'm getting the following exception when I try to open the MISO PSSE-30 power model:

Traceback (most recent call last):
  File "C:\gridcal\venv\lib\site-packages\GridCal\Engine\IO\file_handler.py", line 323, in run
    progress_func=self.progress_signal.emit)
  File "C:\gridcal\venv\lib\site-packages\GridCal\Engine\IO\file_handler.py", line 157, in open
    parser = PSSeParser(self.file_name)
  File "C:\gridcal\venv\lib\site-packages\GridCal\Engine\IO\psse_parser.py", line 2007, in __init__
    self.pss_grid, logs = self.parse_psse()
  File "C:\gridcal\venv\lib\site-packages\GridCal\Engine\IO\psse_parser.py", line 2068, in parse_psse
    sections, sections_dict = self.read_and_split()
  File "C:\gridcal\venv\lib\site-packages\GridCal\Engine\IO\psse_parser.py", line 2045, in read_and_split
    name = re.search(str_a + '(.*)' + str_b, srch).group(1).strip()
AttributeError: 'NoneType' object has no attribute 'group'

Could you please help? 

Thanks

SanPen, 2020-05-26:
Hi!

An error in the parser is a bad thing...

Have you tried with other versions of the same file? Say version 33?

I guess that you cannot send the file for me to see what is going on, but if you can I may be able to help. Otherwise, the best you can do is to debug the process and see why is there a problem.

raw files are notorious for their very low quality. Even some raw files generated with PSS/e (i.e. PSS/E 32) have bugs against their own specification. I think it is easier to do it right that what they do.

rafikfahem, 2020-05-26:
Hi Santiago. Thanks for the prompt response!

Sorry I'm unable to share the file and MISO only provides version 30.

I'm currently debugging the process to see where the problem is.

Thanks

closed 2020-06-10 by SanPen