SanPen/VeraGrid #124: Python 3.8 TypeError: 'Shiboken.ObjectType' object is not iterable

bergkvist, 2020-06-07:
Trying to start GridCal in Python 3.8 on Manjaro Linux (rolling release):
```sh
python3 -c "from GridCal.ExecuteGridCal import run; run()"
```
Throws the exception: "TypeError: 'Shiboken.ObjectType' object is not iterable"

<details><summary>Stack trace</summary>

```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/tobias/.local/lib/python3.8/site-packages/GridCal/ExecuteGridCal.py", line 23, in <module>
    from GridCal.Gui.Main.GridCalMain import run
  File "/home/tobias/.local/lib/python3.8/site-packages/GridCal/Gui/Main/GridCalMain.py", line 20, in <module>
    from GridCal.Gui.GridEditorWidget import *
  File "/home/tobias/.local/lib/python3.8/site-packages/GridCal/Gui/GridEditorWidget/__init__.py", line 15, in <module>
    from GridCal.Gui.GridEditorWidget.editor import *
  File "/home/tobias/.local/lib/python3.8/site-packages/GridCal/Gui/GridEditorWidget/editor.py", line 23, in <module>
    from GridCal.Engine.Core.multi_circuit import MultiCircuit
  File "/home/tobias/.local/lib/python3.8/site-packages/GridCal/Engine/__init__.py", line 17, in <module>
    from GridCal.Engine.basic_structures import *
  File "/home/tobias/.local/lib/python3.8/site-packages/GridCal/Engine/basic_structures.py", line 23, in <module>
    from GridCal.Engine.plot_config import LINEWIDTH, plt
  File "/home/tobias/.local/lib/python3.8/site-packages/GridCal/Engine/plot_config.py", line 20, in <module>
    from matplotlib import pyplot as plt
  File "/usr/lib/python3.8/site-packages/matplotlib/pyplot.py", line 2349, in <module>
    switch_backend(rcParams["backend"])
  File "/usr/lib/python3.8/site-packages/matplotlib/pyplot.py", line 221, in switch_backend
    backend_mod = importlib.import_module(backend_name)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/usr/lib/python3.8/site-packages/matplotlib/backends/backend_qt5agg.py", line 11, in <module>
    from .backend_qt5 import (
  File "/usr/lib/python3.8/site-packages/matplotlib/backends/backend_qt5.py", line 15, in <module>
    import matplotlib.backends.qt_editor.figureoptions as figureoptions
  File "/usr/lib/python3.8/site-packages/matplotlib/backends/qt_editor/figureoptions.py", line 14, in <module>
    from matplotlib.backends.qt_editor import _formlayout
  File "/usr/lib/python3.8/site-packages/matplotlib/backends/qt_editor/_formlayout.py", line 103, in <module>
    class ColorLayout(QtWidgets.QHBoxLayout):
TypeError: 'Shiboken.ObjectType' object is not iterable
```

</details>

<details><summary>Installation logs (pip install GridCal)</summary>

```
Defaulting to user installation because normal site-packages is not writeable
Collecting GridCal
  Downloading GridCal-3.7.0.tar.gz (12.6 MB)
     |████████████████████████████████| 12.6 MB 12.0 MB/s 
Collecting PySide2==5.13
  Downloading PySide2-5.13.0-5.13.0-cp35.cp36.cp37-abi3-manylinux1_x86_64.whl (152.9 MB)
     |████████████████████████████████| 152.9 MB 23 kB/s 
Requirement already satisfied: numpy>=1.14.0 in /usr/lib/python3.8/site-packages (from GridCal) (1.18.2)
Requirement already satisfied: scipy>=1.0.0 in /usr/lib/python3.8/site-packages (from GridCal) (1.4.1)
Requirement already satisfied: networkx>=2.1 in /usr/lib/python3.8/site-packages (from GridCal) (2.4)
Requirement already satisfied: pandas>=0.22 in /usr/lib/python3.8/site-packages (from GridCal) (1.0.3)
Collecting xlwt>=1.3.0
  Downloading xlwt-1.3.0-py2.py3-none-any.whl (99 kB)
     |████████████████████████████████| 99 kB 2.4 MB/s 
Requirement already satisfied: xlrd>=1.1.0 in /usr/lib/python3.8/site-packages (from GridCal) (1.2.0)
Requirement already satisfied: matplotlib>=2.1.1 in /usr/lib/python3.8/site-packages (from GridCal) (3.1.2)
Requirement already satisfied: qtconsole>=4.3.1 in /usr/lib/python3.8/site-packages (from GridCal) (4.6.0)
Collecting pyDOE>=0.3.8
  Downloading pyDOE-0.3.8.zip (22 kB)
Collecting pySOT>=0.2.1
  Downloading pySOT-0.2.3.tar.gz (217 kB)
     |████████████████████████████████| 217 kB 10.1 MB/s 
Requirement already satisfied: openpyxl>=2.4.9 in /usr/lib/python3.8/site-packages (from GridCal) (3.0.3)
[... 56 lines cut ...]
Requirement already satisfied: jinja2>=2.9 in /usr/lib/python3.8/site-packages (from folium->GridCal) (2.11.1)
Requirement already satisfied: pyzmq>=13 in /usr/lib/python3.8/site-packages (from jupyter-client>=4.1->qtconsole>=4.3.1->GridCal) (18.1.1)
Requirement already satisfied: tornado>=4.1 in /usr/lib/python3.8/site-packages (from jupyter-client>=4.1->qtconsole>=4.3.1->GridCal) (6.0.3)
Requirement already satisfied: backcall in /usr/lib/python3.8/site-packages (from ipython->smopy>=0.0.6->GridCal) (0.1.0)
Requirement already satisfied: pickleshare in /usr/lib/python3.8/site-packages (from ipython->smopy>=0.0.6->GridCal) (0.7.5)
Requirement already satisfied: jedi>=0.10 in /usr/lib/python3.8/site-packages (from ipython->smopy>=0.0.6->GridCal) (0.15.2)
Requirement already satisfied: prompt-toolkit!=3.0.0,!=3.0.1,<3.1.0,>=2.0.0 in /usr/lib/python3.8/site-packages (from ipython->smopy>=0.0.6->GridCal) (3.0.2)
Requirement already satisfied: pexpect; sys_platform != "win32" in /usr/lib/python3.8/site-packages (from ipython->smopy>=0.0.6->GridCal) (4.7.0)
Requirement already satisfied: idna>=2.5 in /usr/lib/python3.8/site-packages (from requests->folium->GridCal) (2.9)
Requirement already satisfied: urllib3>=1.21.1 in /usr/lib/python3.8/site-packages (from requests->folium->GridCal) (1.25.8)
Requirement already satisfied: MarkupSafe>=0.23 in /usr/lib/python3.8/site-packages (from jinja2>=2.9->folium->GridCal) (1.1.1)
Requirement already satisfied: parso>=0.5.2 in /usr/lib/python3.8/site-packages (from jedi>=0.10->ipython->smopy>=0.0.6->GridCal) (0.5.2)
Requirement already satisfied: ptyprocess>=0.5 in /usr/lib/python3.8/site-packages (from pexpect; sys_platform != "win32"->ipython->smopy>=0.0.6->GridCal) (0.6.0)
Installing collected packages: shiboken2, PySide2, xlwt, pyDOE, pyDOE2, POAP, dill, pySOT, smopy, geographiclib, geopy, h5py, branca, folium, GridCal
    Running setup.py install for pyDOE ... done
    Running setup.py install for pyDOE2 ... done
    Running setup.py install for dill ... done
    Running setup.py install for pySOT ... done
    Running setup.py install for GridCal ... done
Successfully installed GridCal-3.7.0 POAP-0.1.26 PySide2-5.13.0 branca-0.4.1 dill-0.3.1.1 folium-0.11.0 geographiclib-1.50 geopy-1.22.0 h5py-2.10.0 pyDOE-0.3.8 pyDOE2-1.3.0 pySOT-0.2.3 shiboken2-5.13.0 smopy-0.0.7 xlwt-1.3.0
```

</details>

It seems to be caused by this issue: https://bugreports.qt.io/browse/PYSIDE-1140, so updating the version of PySide should likely fix the problem.

SanPen, 2020-06-07:
Hi Tobias, 

PySide2 and pyqtconsole have been recently shipping with a horrible quality.
Therefore pip installing GridCal is not as straight forward as it used to be.

I can assure you that I am currently running and developing GridCal under Ubuntu 20.0 with Python 3.8. My package list is the following:

```
apturl==0.5.2
asgiref==3.2.7
atomicwrites==1.1.5
attrs==19.3.0
backcall==0.1.0
bcrypt==3.1.7
beautifulsoup4==4.8.2
blinker==1.4
branca==0.4.1
Brlapi==0.7.0
certifi==2019.11.28
chardet==3.0.4
command-not-found==0.3
cryptography==2.8
cupshelpers==1.0
cycler==0.10.0
dbus-python==1.2.16
decorator==4.4.2
defer==1.0.6
dill==0.3.1.1
distro==1.4.0
distro-info===0.23ubuntu1
Django==3.0.6
django-embed-video==1.3.2
duplicity==0.8.12.0
easy-thumbnails==2.7
entrypoints==0.3
et-xmlfile==1.0.1
fasteners==0.14.1
folium==0.11.0
future==0.18.2
geographiclib==1.50
geopy==1.20.0
h5py==2.10.0
html5lib==1.0.1
httplib2==0.14.0
idna==2.8
importlib-metadata==1.5.0
ipykernel==5.3.0
ipython==7.14.0
ipython-genutils==0.2.0
jdcal==1.0
jedi==0.17.0
Jinja2==2.11.2
joblib==0.14.0
jupyter-client==6.1.3
jupyter-core==4.6.3
keyring==18.0.1
keyrings.alt==3.4.0
kiwisolver==1.0.1
language-selector==0.1
launchpadlib==1.10.13
lazr.restfulclient==0.14.2
lazr.uri==1.0.3
llvmlite==0.31.0
lockfile==0.12.2
louis==3.12.0
lxml==4.5.0
macaroonbakery==1.3.1
Mako==1.1.0
MarkupSafe==1.1.0
matplotlib==3.1.2
meson==0.53.2
monotonic==1.5
more-itertools==4.2.0
netifaces==0.10.4
networkx==2.4
nose==1.3.7
numba==0.48.0
numexpr==2.7.1
numpy==1.17.4
oauthlib==3.1.0
olefile==0.46
openpyxl==3.0.3
packaging==20.3
pandas==0.25.3
paperclip==2.2.5
paramiko==2.6.0
parso==0.7.0
pexpect==4.6.0
pickleshare==0.7.5
Pillow==7.0.0
pluggy==0.13.0
POAP==0.1.26
prompt-toolkit==3.0.5
protobuf==3.6.1
psutil==5.5.1
PuLP==2.1
py==1.8.1
pybind11==2.5.0
pycairo==1.16.2
pycrypto==2.6.1
pycups==1.9.73
pyDOE==0.3.8
pyDOE2==1.3.0
Pygments==2.3.1
PyGObject==3.36.0
PyJWT==1.7.1
pymacaroons==0.13.0
PyNaCl==1.3.0
pyparsing==2.4.6
pyRFC3339==1.1
PySide2==5.15.0
pySOT==0.2.3
pytest==4.6.9
python-apt==2.0.0
python-dateutil==2.7.3
python-debian===0.1.36ubuntu1
pytz==2019.3
pyxdg==0.26
PyYAML==5.3.1
pyzmq==19.0.1
qtconsole==4.3.1
reportlab==3.5.34
requests==2.22.0
requests-oauthlib==1.0.0
requests-unixsocket==0.2.0
scikit-learn==0.22.2.post1
scipy==1.3.3
SecretStorage==2.3.1
shiboken2==5.15.0
simplejson==3.16.0
six==1.14.0
smopy==0.0.7
soupsieve==1.9.5
sqlparse==0.3.1
systemd-python==234
tables==3.6.1
tornado==6.0.4
traitlets==4.3.3
ubuntu-advantage-tools==20.3
ubuntu-drivers-common==0.0.0
ufw==0.36
unattended-upgrades==0.1
urllib3==1.25.8
usb-creator==0.3.7
vboxapi==1.0
wadllib==1.3.3
wcwidth==0.1.8
webencodings==0.5.1
xkit==0.0.0
xlrd==1.1.0
XlsxWriter==1.2.9
xlwt==1.3.0
zipp==1.0.0
```

Please tell me if by installing the same shiboken package version you manage to get it working.

BR,
Santiago

bergkvist, 2020-06-07:
Thanks Santiago!

I just tried:
```
pip install PySide2==5.15.0
```

Logs (notice that pip complains about gridcal 3.7.0 being incompatible with PySide2==5.15.0)
```
Defaulting to user installation because normal site-packages is not writeable
Collecting PySide2==5.15.0
  Downloading PySide2-5.15.0-5.15.0-cp35.cp36.cp37.cp38-abi3-manylinux1_x86_64.whl (170.8 MB)
     |████████████████████████████████| 170.8 MB 8.3 kB/s 
Collecting shiboken2==5.15.0
  Downloading shiboken2-5.15.0-5.15.0-cp35.cp36.cp37.cp38-abi3-manylinux1_x86_64.whl (856 kB)
     |████████████████████████████████| 856 kB 10.4 MB/s 
ERROR: gridcal 3.7.0 has requirement PySide2==5.13, but you'll have pyside2 5.15.0 which is incompatible.
Installing collected packages: shiboken2, PySide2
  Attempting uninstall: shiboken2
    Found existing installation: shiboken2 5.13.0
    Uninstalling shiboken2-5.13.0:
      Successfully uninstalled shiboken2-5.13.0
  Attempting uninstall: PySide2
    Found existing installation: PySide2 5.13.0
    Uninstalling PySide2-5.13.0:
      Successfully uninstalled PySide2-5.13.0
Successfully installed PySide2-5.15.0 shiboken2-5.15.0
```

Ignoring the error, the GridCal GUI successfully starts with:
```
python3 -c "from GridCal.ExecuteGridCal import run; run()"
```

Logs:
```
KLU failed
Pardiso failed
UmfPack failed
Falling back to Blas/Lapack
Using Blas/Lapack
```

What is KLU, Pardiso and UmfPack? Should I worry that these "failed"?

SanPen, 2020-06-07:
Hi, 

Not at all, this is because you may install other linear algebra packages (with quite some effort) 
KLU is a linear algebra library done ex-profeso for electrical circuits.
My tests show that there is no real advantage over the regular BLAS/LAPACK libraries that ship with scipy.

TL;DR: no need to worry

SanPen, 2020-06-07:
About PySide2, there is a [bug](https://bugreports.qt.io/browse/PYSIDE-1171) that I reported and has not been resolved yet.

If you need to develop the GUI, install PySide2 5.13, otherwise the newer versions should work.

closed 2020-06-10 by SanPen