SanPen/VeraGrid #125: GridCal 4.0.0

einarvh, 2020-06-29:
Hi!

The GridCal documentation refers to GridCal 4.0.0 several times, I am wondering when this version will be available for download? There are several functions there that would be very useful for a project I am working on!

SanPen, 2020-06-29:
Hi!, 

The release will take some time still since I cannot devote as much time as before...

However you can clone and install the master branch and most of the features will work.

If you tell me which are the things that you are interested in, I can tell you the status.

best regards,
Santiago

einarvh, 2020-06-30:
Hi!

Thanks for the quick reply! I was mostly interested in the changes in how the devices are managed.

SanPen, 2020-06-30:
Hi, 

That part should be working. Not sure about what you need, but at least what I intended is done in that regard.

SanPen, 2021-01-13:
Hi, GridCal 4.0.0 Alpha versions are published!
https://pypi.org/project/GridCal/4.0.0a5/

closed 2021-01-19 by SanPen