SanPen/VeraGrid #126: Issue with Time Series Power Flow

sergio-dorado, 2020-09-02:
Hi, 

I am trying to run time-series power flow computations for several systems imported from PSS/E by reading `.raw` files directly. I'm doing it using the following commands:

```python
file_handler = FileOpen(os.path.abspath(os.path.join("tests", 
                                                     "model_files", 
                                                     "psse", 
                                                     "Seven_Bus", 
                                                     "SevenBus_Base_Case.raw")))

grid_raw = file_handler.open()
```

Following the [tutorial](https://github.com/SanPen/GridCal/blob/master/src/Tutorials/defining_a_grid_from_scratch_with_profiles.py), the next step is to define a grid time profile which I do after reading some data I have in a data frame. 

```python
grid_raw.time_profile = df_dict['CAPITL'].index
```
Next, I define some profiles. For simplicity, I set them to be the same original power flow settings:

```python
for gen in grid_raw.get_generators():
    gen.P_prof = gen.P * np.ones(shape = (len(grid_raw.time_profile), ))
    gen.Vset_prof = gen.Vset * np.ones(shape = (len(grid_raw.time_profile), ))

for load in grid_raw.get_loads():
    load.P_prof = load.P * np.ones(shape = (len(grid_raw.time_profile), ))
    load.Q_prof = load.Q * np.ones(shape = (len(grid_raw.time_profile), ))
```

After defining the power flow settings, it is possible to run a power flow calculation. However, for time-series power flow, the program fails since some attributes are missing in the grid model.

```python
# Power flow options
options = PowerFlowOptions(SolverType.NR,
                           verbose=True,
                           initialize_with_existing_solution = False,
                           multi_core = False,
                           tolerance = 1e-9,
                          max_iter = 99,
                          control_q = True)

pf = PowerFlowDriver(grid_raw, options)
pf.run()

ts = TimeSeries(grid = grid_raw, 
                options = options, 
                opf_time_series_results = None, 
                start_ = 0, 
                end_ = None)

ts.run()

```

More specifically, this is the error that prompts:

[image]

I have found that an ad-hoc solution for the system I am working with is to change the properties of the elements that have a profile from `None` to arrays of zeros depending on the field. 

```python
n_points_ts = len(grid_raw.time_profile)

for load in grid_raw.get_loads():
    load.B_prof = np.zeros(shape = (n_points_ts, ))
    load.Cost_prof = np.zeros(shape = (n_points_ts, ))
    load.G_prof = np.zeros(shape = (n_points_ts, ))
    load.Ii_prof = np.zeros(shape = (n_points_ts, ))
    load.Ir_prof = np.zeros(shape = (n_points_ts, ))
    load.P_prof = load.P * np.ones(shape = (n_points_ts, ))
    load.Q_prof = load.Q * np.zeros(shape = (n_points_ts, ))
    load.active_prof = np.asarray([True for _ in range(n_points_ts)])
    
for gen in grid_raw.get_generators():
    gen.Cost_prof = gen.Cost * np.ones(shape = (n_points_ts,))
    gen.P_prof = gen.P * np.ones(shape = (n_points_ts,))
    gen.Pf_prof = gen.Pf * np.ones(shape = (n_points_ts,))
    gen.Vset_prof = gen.Vset * np.ones(shape = (n_points_ts,))
    gen.active_prof = np.asarray([True for _ in range(n_points_ts)])

for branch in grid_raw.get_branches():
    branch.Cost_prof = branch.Cost * np.ones(shape = (n_points_ts, ))
    branch.rate_prof = branch.rate * np.ones(shape = (n_points_ts, ))
    branch.temp_oper_prof = branch.temp_oper * np.ones(shape = (n_points_ts, ))
    branch.active_prof = np.asarray([True for _ in range(n_points_ts)])
    
for bus in grid_raw.get_buses():
    bus.active_prof = np.asarray([True for _ in range(n_points_ts)])
```

I would like to know what your thoughts are on how to add this piece of code to the `PSSeParser` class. According to the `elements_with_profile` method of the `MultiCircuit` class, I am missing a few items (Lines, DC Lines, VSC, among others) but apparently there isn't a getter for those yet. 

Another problem is related to the reactive power control in the power flow options for the time-series computations. The `control_q` setting cannot be changed to `ReactivePowerControlMode.Direct`. I get the following error:

[image]

For `ReactivePowerControlMode.Iterative`, this pops up:

[image]

I have no issues running the time-series power flow with `ReactivePowerControlMode.NoControl` and `control_q = True`, `control_q = False`. I'd appreciate any input into how this can be properly set up for such computations.

Kind regards,

Sergio

SanPen, 2020-09-02:
Hi Sergio, 

Plenty to have a look at!

First, which version are you using? I'd assume it is 3.7.1, right?

I would use the `format_profiles` property of the `MultiCircuit` to format all the profiles at once. This avoids the more than likely issue that some of the profile variables are `None`. In your case, you should use `grid_raw.format_profiles(df_dict["CAPITL"].index)`. This avoids having to set up everything manually. Certainly it is a point to improve in the tutorial.

Regarding the additional devices, in the `devel` branch of GridCal you may find that I've bee working on that and those  devices you mention are there already. But!! I have been unable to come up with a workable AC-DC power flow method. The best candidate so far is outlined at "Flexible General Branch Unified Power Flow Algorithm for Future Flexible AC/DC Networks - Abraham Álvarez and Behzad Kazemtabrizi" but I must admit that I am struggling there.

Regarding the Q control, please try again after using the function `format_profiles`. It may not be the case, but to be sure.

I am not sure about the time-series initialization in the PSSeParser, since RAW files are not time based.
Anyway, If you consider any contribution, I'll go in the development version for sure, and that one has suffered quite some changes.

Best regards,
Santiago

sergio-dorado, 2020-09-02:
Hi Santiago,

Thanks for your answer. I'm using the development version. The hash is

```
564ffdd8c52e5d64fecc22ac042f14e118489385
```

The `format_profiles` function is working fine! Thanks! Do you want me to gather the code and include it in the `Tutorials` with a `.raw` file? I think that by doing so it won't be necessary to do any changes in the PSSeParser since it would be clear that `format_profiles` has to be used in advance.

Regarding Q control, the time-series power flow computation:

- doesn't run for `ReactivePowerControlMode.Direct`;
- runs for `ReactivePowerControlMode.Iterative` but the bus voltages are pretty low;
- works for `ReactivePowerControlMode.NoControl`, `control_q = True` and `control_q = False`.

To give you a better idea, in the model I am working with I get the following results with PSS/E for bus magnitudes:

```python
[1.0563983 , 1.06825282, 1.00755608, 1.03889374, 0.99905094, 1.05357656, 1.08970511, 1.        , 1.        , 1.02      ])
```

I get the same result with `ReactivePowerControlMode.Direct` and the boolean options for `control_q` for a single power flow. For the time-series, I get the same except for `ReactivePowerControlMode.Iterative` where the output is:

```python
[0.8330635 , 0.7106643 , 0.63301409, 0.82252445, 0.6013843 , 0.72817824, 1.00700405, 0.65003991, 0.68342058, 1.02      ]

```

SanPen, 2020-09-03:
Hi Sergio, 

So you're on the very latest (but not greatest) version.That is fine since we can fix things.

If you observe the tests folder, there are a number of tests, and if you run them in bulk (you can do that with Pycharm) you'll see that about half fail. The Q control is one of those. Clearly we may try to fix those. If you help on that it would be awesome.

The major change is that before, the control variables were passed to the process in a hacky way, and to improve that and other things, I vastly changed the SnapShotCircuit and TimeSeriesCircuit classes. So it should be related to the data stored there and how it is dealt with.

If you haven't noticed, the data flow in GridCal is:
Objects -> Arrays -> Arrays per island -> Outer/Inner loop control -> Numerical methods

It is in the outer/inner control part where the reactive power is managed.

Regarding, the tutorial, yes please, make a pull request and I'll add it.

sergio-dorado, 2020-09-22:
Hi Santiago,

Sorry for the late reply. I was rushing with a submission... 

Could you please share with me the hash of the greatest GridCal version? I would like to freeze the stuff I am working on with a more-or-less stable release.

As soon as I complete the tutorial I'll make a pull request. I'd have to change the test system since I am forcing Q limits on some generators and that won't work for the time being.

I could try to give you a hand with the Q control! Can I email you about it? I'd need a deeper explanation. 

Thanks,

Sergio

SanPen, 2020-09-23:
Hi, 

The last stable release is [3.7.0](https://github.com/SanPen/GridCal/releases/tag/3.7.0), hash: 78f5f5706f8205793a67f74108ee7b31ece852a9 I believe.

But if you want to work on something, or make a pull request that would actually be accepted it'll have to be to de "devel" branch since there are so many changes.

You can email me or we can have a chat via google hangouts.

Best regards,
Santiago

closed 2020-10-28 by SanPen