SanPen/VeraGrid #127: Loading a .raw document from the script

MathieuGR, 2020-09-15:
Good morning,

I currently have a problem using GridCal.
I have a .raw document (from PSS/E) describing a grid.
I would like to load this file, do its power flow and get the results has matrices, all of that from a python script using GridCal as a library.
In a second step I would also like to change some of the parameters in order to to an iterative power flow.

I have already check that I can do the power flow of my .raw file with the graphical interface.

Is there any example of how to load a file in a GridCal script?

Thank you for your help

SanPen, 2020-09-15:
Hi Mathieu,

You can do the following:

```
from GridCal.Engine import *

fname = 'path_to_my_file.raw' 
main_circuit = FileOpen(fname).open()
```

Additionally, you can check out the [power flow test](https://github.com/SanPen/GridCal/blob/master/src/tests/test_power_flow.py).

Best regards,
Santiago

MathieuGR, 2020-09-21:
Hello Santiago and thank you for the advice, 
Currently I am using the following code:

`from GridCal.Engine import *

fname = 'kundur_two_area_v33.raw'
grid = FileOpen(fname).open()

options = PowerFlowOptions(SolverType.NR, verbose=False)
power_flow = PowerFlowDriver(grid, options)
power_flow.run()

print('\n\n', grid.name)
print('\t|V|:', abs(power_flow.results.voltage))
print('\t|Sbranch|:', abs(power_flow.results.Sbranch))
print('\t|loading|:', abs(power_flow.results.loading) * 100)
print('\terr:', power_flow.results.error)
print('\tConv:', power_flow.results.converged)

grid.plot_graph()
plt.show() `

It allows me to run the power flow of my grid. However I don't manage to get all the results I want. In particular I don't have the voltage angles of the buses, neither the reactive power of the generator (and the active power of the slack-generator).
Is there a documentation to get all of the functions for calling the results?

Finally is there a solution for changing some of the parameters of my grid directly in the script once I have loaded it? (by example the active or reactive power absorbed by the loads.)

Thank you for your help,
Mathieu

SanPen, 2020-09-21:
Hi, 

Indeed you can get the results you want.

GridCal computes the magnitudes in complex format, hence `voltage` contains the complex voltage. The same with the nodal power `Sbus`.

So, to get the voltages angles:
```
angles = np.angle(power_flow.results.voltage)
```

To get the reactive power at the generation buses:
```
Qgen = power_flow.results.Sbus.imag[generator_bus_indices]
```

To get the slack power:
```
Sslack = power_flow.results.Sbus[slack_bus_indices]
```

To change the values of the grid objects, you simply need to access to the values and change them, for instance:

```
# Set the active power of the first load to 10 MW
grid.loads[0].P = 10
```

The internals are mostly documented. See the load [here](https://gridcal.readthedocs.io/en/latest/api/auto/GridCal.Engine.Devices.html#module-GridCal.Engine.Devices.load) or the generator [here](https://gridcal.readthedocs.io/en/latest/api/auto/GridCal.Engine.Devices.html#module-GridCal.Engine.Devices.generator).

Of course there is a more advanced way to modify values if you are interested in doing faster changes, if you need to, let me know. 

Best regards,
Santiago

MathieuGR, 2020-09-22:
Hello,

The code for getting the results work well, thank you.

However I don't manage to change the power of the load.
I use this code:
```fname = 'kundur_two_area_v33.raw'
grid = FileOpen(fname).open()
grid.load[0].P = 1000
```

and get this error message:
```
KLU failed
Pardiso failed
UmfPack failed
Falling back to Blas/Lapack
Using Blas/Lapack
Traceback (most recent call last):
  File "C:/Users/m.robin/PycharmProjects/GridCal/reading.py", line 6, in <module>
    grid.load[0].P = 10
AttributeError: 'MultiCircuit' object has no attribute 'load'
```

The document I load (''kundur_two_area_v33.raw') define the loads as follow:
```
0 / END OF BUS DATA, BEGIN LOAD DATA
     7,'1 ',1,   1,   1,  967.000,   -100.000,     0.000,     0.000,     0.000,     0.000,   1,1,0
     9,'1 ',1,   1,   1,  1767.000,   -250.000,     0.000,     0.000,     0.000,     0.000,   1,1,0
0 / END OF LOAD DATA, BEGIN FIXED SHUNT DATA
```
Best regards,
Mathieu

SanPen, 2020-09-22:
Hi, 

It is `loads` in plural:
```
grid = FileOpen(fname).open()
grid.loads[0].P = 1000
```

Tell me if it works.

BR,
Santiago

MathieuGR, 2020-09-22:
Hello, 

actualy I tried first with loads but tried in singular after. 
I get the same error message with the plural:
```
Traceback (most recent call last):
  File "C:/Users/m.robin/PycharmProjects/GridCal/reading.py", line 5, in <module>
    grid.loads[0].P = 1000
AttributeError: 'MultiCircuit' object has no attribute 'loads'
```

BR,
Mathieu

SanPen, 2020-09-22:
Totally my fault!

it is `grid.get_loads()` to get all the loads, or alternatively `grid.buses[0].loads[0]` to access the load devices located in a bus.

GridCal stores the shunt elements (loads, generators, batteries, etc...) inside their bus object.

MathieuGR, 2020-09-30:
Hello again,
Thank you this two commands worked well.

However is there an equivalent command to get the list of all the buses and all the branches ? Because it looks like `grid.get_buses()` or  `grid.get_branches()`  doesn't work.

Additionally is there a way to save my modified .raw document (with new values of loads by example)? `save(file_name)` doesn't work.

BR,
Mathieu

SanPen, 2020-09-30:
Hi, 

Only in the case of buses and branches (because they are best assimilated to the elements of a graph) they are stored directly in the circuit. So you can access them via `grid.buses`or `grid.branches`. They are list objects. I'd be surprise if the getters for buses and branches did not work since they are the most simple functions.

To save a file, you do 'FileSave(grid, 'my_filename.gridcal').save()`.

Why not, simply `grid.save(''filename.gridcal')` ? It was like that and it was a mess to properly maintain.

There is a variety of formats supported for saving; .gridcal, .xlsx, .json...but not .raw. I have not implemented a raw file export, mainly because the .raw files are a terrible mess. So much, that even PSS/e gets it wrong some times: i.e. version 32 are wrong by their own file specification. But if you're brave enough to try I may be of assistance.

BR,
Santiago

closed 2020-10-28 by SanPen