SanPen/VeraGrid #128: academic publications

robbiemorrison, 2021-01-07:
Could the developers provide a list of academic publications (if any)?  Perhaps to the bottom of the `readme.md` or in the `readthedocs`.

If there are suitable publications, I'll add the project to wikipedia (see [talk&nbsp;page](https://en.wikipedia.org/wiki/Talk:Open_energy_system_models#Further_models) and [article](https://en.wikipedia.org/wiki/Open_energy_system_models#Electricity_sector_models))

Thanks!

SanPen, 2021-01-07:
Hi Robbie, 

Thanks for including the program in those pages!

There are no publications about GridCal per se. 
Since I personally have very little incentives to publish, I just put the time in the coding part. 
However I welcome whoever wants to help to change that. I did register the DOI some years ago but that is it.

There are tons of articles and books checked to produce GridCal, and when improving the "[Theory behind GridCal](https://gridcal.readthedocs.io/en/latest/theory/models.html)" section we try to mention as many references as it makes sense.

robbiemorrison, 2021-01-07:
Hi Santiago.  A reliable secondary source in normally needed before material can be included in Wikipedia.  With academic publications being particularly attractive of course.  Nonetheless I&nbsp;looked at the background in your documentation and it looks sufficiently comprehensive.  So I&nbsp;think I&nbsp;can go with that and duly edit Wikipedia.  Hopefully, other people will use GridCal and publish their experiences academically in due course?  Thanks for the reply.  From my perspective, you can mark this issue resolved.&nbsp;R.

closed 2021-01-08 by SanPen