SanPen/VeraGrid #129: run error ‘optimal power flow’

lym8713, 2021-01-28:
input file:Lynn 5 Bus pv.gridcal
error msg:
Traceback (most recent call last):
  File "/Users/liushuo/Documents/GridCal各版本备份/GridCal-master/src/GridCal/Engine/Simulations/OPF/opf_driver.py", line 171, in run
    self.opf()
  File "/Users/liushuo/Documents/GridCal各版本备份/GridCal-master/src/GridCal/Engine/Simulations/OPF/opf_driver.py", line 120, in opf
    problem = OpfDc(numerical_circuit=numerical_circuit, solver=self.options.mip_solver)
  File "/Users/liushuo/Documents/GridCal各版本备份/GridCal-master/src/GridCal/Engine/Simulations/OPF/dc_opf.py", line 186, in __init__
    Opf.__init__(self, numerical_circuit=numerical_circuit, solver=solver)
  File "/Users/liushuo/Documents/GridCal各版本备份/GridCal-master/src/GridCal/Engine/Simulations/OPF/opf_templates.py", line 45, in __init__
    self.problem = self.formulate()
  File "/Users/liushuo/Documents/GridCal各版本备份/GridCal-master/src/GridCal/Engine/Simulations/OPF/dc_opf.py", line 242, in formulate
    cost_g, cost_b, cost_l, cost_br)
  File "/Users/liushuo/Documents/GridCal各版本备份/GridCal-master/src/GridCal/Engine/Simulations/OPF/dc_opf.py", line 41, in add_objective_function
    f_obj = lpSum(cost_g * Pg)
  File "/Users/liushuo/Documents/GridCal各版本备份/GridCal-master/src/GridCal/ThirdParty/pulp/pulp.py", line 2109, in lpSum
    return LpAffineExpression().addInPlace(vector)
  File "/Users/liushuo/Documents/GridCal各版本备份/GridCal-master/src/GridCal/ThirdParty/pulp/pulp.py", line 733, in addInPlace
    if other == 0 or other is None:
ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()

SanPen, 2021-01-28:
Hi!, 

Which version are you using?

Best regards,
Santiago

lym8713, 2021-01-28:
Hi, SanPen
I use  GridCal 4.0
All Best!
Lym8713

SanPen, 2021-01-28:
Hi Lym, 

I have just tried ant it seems to work on the latest version.

So how do you have it installed? if it is from the master repository you may update now.

If you have it installed with pip, you may upgrade to the latest alpha version 4.0.0a8 with 

`pip install GridCal=4.0.0a8 --upgrade --upgrade-strategy only-if-needed `

Tell me if you manage.

Best regards,
Santiago

lym8713, 2021-01-29:
Hi,SanPen
My problem has been solved by your update！
Thank you for your help!
All Best!
Lym8713

closed 2021-01-30 by SanPen