SanPen/VeraGrid #131: Issue - Overhead lines

SodinD, 2021-03-31:
After following the tutorial on the Distribution Grid we have an issue at Step 9: Defining Wires and Overhead Lines.
When we try to perform  point 7: (Select the wire ‘AWG SLD’, highlight it and click on the ‘+’ sign on the ‘Wire composition’ section below:), nothing happens and we get the following message in comand line.

Traceback (most recent call last):
  File "C:\GridCal\python\lib\site-packages\GridCal\Gui\TowerBuilder\LineBuilderDialogue.py", line 131, in add_wire_to_tower
    selected_wire = self.wires_table.wires[sel_idx].copy()
  File "C:\GridCal\python\lib\site-packages\GridCal\Engine\Devices\wire.py", line 60, in copy
    return Wire(self.name, self.gmr, self.r, self.x, self.max_current)
  File "C:\GridCal\python\lib\site-packages\GridCal\Engine\Devices\wire.py", line 34, in __init__
    EditableDevice.__init__(self,
  File "C:\GridCal\python\lib\site-packages\GridCal\Engine\Devices\editable_device.py", line 66, in __init__
    self.idtag = idtag.replace('_', '').replace('-', '')
AttributeError: 'float' object has no attribute 'replace'

SanPen, 2021-03-31:
Hi!

Thanks for reporting.

Could you please tell me the version of the program you are using?
Also, how did you install it?
Linux? windows? osx?

So far, what you mention works in v4.0.0 for me, but it may be that it does not in windows due to a certain dependency update. That has happened in the past :/

If you call `python -m pip freeze` with the python executable that you are using it will display a list of the packages and their versions. That is usually good to have as well.

BR,
Santiago

SanPen, 2021-03-31:
Hi, 

The excel failure was due to the xlrd package.
The wire failure was a bug.
I have just fixed those issues with an update to v4.0.1

If you installed the windows standalone version, I encourage you to download and install again due to the changes that the xlrd package thing imposed.

Tell me if you succeed.

BR,
Santiago

SodinD, 2021-04-01:
Hey,

Thanks for the prompt response.
Both features work for me now as well, however the point 5 (Any load profile can be selected. For example, click on ‘USA_AL_Dothan.Muni.AP.7222268_TMY3_BASE(kW)’. Then select the ‘Plot’ tab to see the load profile in kW for January 2018.) of Step 10: importing Load profiles, still does not show the preview of the consumption. I know that this is a minor issue, just wanted to report it anyway.

BR, Denis

SanPen, 2021-04-01:
Hi Denis, 

That is because you need to double click on the series. Then it will plot (al least for me)

BR,
Santiago

SodinD, 2021-04-01:
Double-click indeed solves the issue also on my side.
Thanks for help.

BR, Denis

closed 2021-04-01 by SodinD