SanPen/VeraGrid #132: Unable to run GridCal after installation

YaeSakura1, 2021-04-02:
Hello,after i successfully installed GridCal,i'm unable to run it using this command: python -c "from GridCal.ExecuteGridCal import run; run()" on my windows 10.what should i do next? thank you!!

(base) C:\Users\Administrator>python -c "from GridCal.ExecuteGridCal import run; run()"
Loading GridCal...
qt.qpa.plugin: Could not load the Qt platform plugin "windows" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: direct2d, minimal, offscreen, windows.

SanPen, 2021-04-02:
Hi!

Thanks for reporting the issue.

How did you install the python environment?
did you use Anaconda or a normal python?
Also, which python version?

Looking on stack overflow I find [this](https://stackoverflow.com/questions/56560515/qt-qpa-plugin-could-not-find-the-qt-platform-plugin-windows-in). Which provides a solution I don't like very much...

If you had a previous python (specially Anaconda) that has pyqt5 there are chances of incompatibility with pyside2 which is the newer but official Qt library.

In short, you can follow the stack overflow solution and see if it works, better, just get the standalone setup.
If you want to have a custom python, I advise to install a vanilla python 3.8 from python.org and install GridCal there.

Let me know if any of this works for you.

BR,
Santiago

YaeSakura1, 2021-04-02:
Thanks for your reply.
i install the GridCal use Anaconda ,the python version is 3.8.5

SanPen, 2021-04-02:
Hi, 

Anaconda has an awful management of Qt. See [what we came up](https://gridcal.readthedocs.io/en/latest/getting_started/install.html#pyside2-vs-pyqt5) last time for it.

My advise is that unless you need some really specific feature from Anaconda, go with a regular python installation. Things will work better. In regular python you can have PyQt5 and PySide2 no problem. In Anaconda that will break things, not sure why.

BR,
Santiago

YaeSakura1, 2021-04-02:
after remove the file qt.conf. i get this result

(base) C:\Users\Administrator>python -c "from GridCal.ExecuteGridCal import run; run()"
Loading GridCal...
Traceback (most recent call last):
  File "C:\Users\Administrator\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 535, in get
    value = obj._trait_values[self.name]
KeyError: 'iopub_socket'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Administrator\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 535, in get
    value = obj._trait_values[self.name]
KeyError: 'iopub_thread'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\Administrator\Anaconda3\lib\site-packages\GridCal\Gui\Main\GridCalMain.py", line 5550, in run
    window = MainGUI(use_native_dialogues=use_native_dialogues)
  File "C:\Users\Administrator\Anaconda3\lib\site-packages\GridCal\Gui\Main\GridCalMain.py", line 313, in __init__
    self.create_console()
  File "C:\Users\Administrator\Anaconda3\lib\site-packages\GridCal\Gui\Main\GridCalMain.py", line 660, in create_console
    self.console = ConsoleWidget(customBanner="GridCal console.\n\n"
  File "C:\Users\Administrator\Anaconda3\lib\site-packages\GridCal\Gui\ConsoleWidget.py", line 21, in __init__
    self.kernel_manager.start_kernel(show_banner=False)
  File "C:\Users\Administrator\Anaconda3\lib\site-packages\ipykernel\inprocess\manager.py", line 46, in start_kernel
    self.kernel = InProcessKernel(parent=self, session=self.session)
  File "C:\Users\Administrator\Anaconda3\lib\site-packages\ipykernel\inprocess\ipkernel.py", line 72, in __init__
    super(InProcessKernel, self).__init__(**traits)
  File "C:\Users\Administrator\Anaconda3\lib\site-packages\ipykernel\ipkernel.py", line 71, in __init__
    self.shell.displayhook.pub_socket = self.iopub_socket
  File "C:\Users\Administrator\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 575, in __get__
    return self.get(obj, cls)
  File "C:\Users\Administrator\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 538, in get
    default = obj.trait_defaults(self.name)
  File "C:\Users\Administrator\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 1578, in trait_defaults
    return self._get_trait_default_generator(names[0])(self)
  File "C:\Users\Administrator\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 975, in __call__
    return self.func(*args, **kwargs)
  File "C:\Users\Administrator\Anaconda3\lib\site-packages\ipykernel\inprocess\ipkernel.py", line 67, in _default_iopub_socket
    return self.iopub_thread.background_socket
  File "C:\Users\Administrator\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 575, in __get__
    return self.get(obj, cls)
  File "C:\Users\Administrator\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 538, in get
    default = obj.trait_defaults(self.name)
  File "C:\Users\Administrator\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 1578, in trait_defaults
    return self._get_trait_default_generator(names[0])(self)
  File "C:\Users\Administrator\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 975, in __call__
    return self.func(*args, **kwargs)
  File "C:\Users\Administrator\Anaconda3\lib\site-packages\ipykernel\inprocess\ipkernel.py", line 59, in _default_iopub_thread
    thread = IOPubThread(self._underlying_iopub_socket)
  File "C:\Users\Administrator\Anaconda3\lib\site-packages\ipykernel\iostream.py", line 71, in __init__
    self._setup_event_pipe()
  File "C:\Users\Administrator\Anaconda3\lib\site-packages\ipykernel\iostream.py", line 90, in _setup_event_pipe
    self._event_puller = ZMQStream(pipe_in, self.io_loop)
  File "C:\Users\Administrator\Anaconda3\lib\site-packages\zmq\eventloop\zmqstream.py", line 127, in __init__
    self._init_io_state()
  File "C:\Users\Administrator\Anaconda3\lib\site-packages\zmq\eventloop\zmqstream.py", line 552, in _init_io_state
    self.io_loop.add_handler(self.socket, self._handle_events, self.io_loop.READ)
  File "C:\Users\Administrator\Anaconda3\lib\site-packages\tornado\platform\asyncio.py", line 100, in add_handler
    self.asyncio_loop.add_reader(fd, self._handle_events, fd, IOLoop.READ)
  File "C:\Users\Administrator\Anaconda3\lib\asyncio\events.py", line 501, in add_reader
    raise NotImplementedError
NotImplementedError

SanPen, 2021-04-02:
As I said, Anaconda does not play nice with Qt.
My honest advice is to get rid of it and install a regular python. I did that years ago and never looked back.

BR,
Santiago

YaeSakura1, 2021-04-02:
you mean i should remove the GridCal in Anaconda and install it a regular python
i will follow your advices
Do I need to delete Anaconda?

SanPen, 2021-04-02:
You can do a number of things:

a) Delete anaconda, install python 3.8 from python.org and install GridCal on it.
b) Keep Anaconda, install python 3.8 from python.org and install GridCal on it. Be aware to not to add it to the path when installing.
c) Keep Anaconda, install the GridCal standalone setup which will come with a functional python distribution. You can later add any other package to that distribution since it is a standard python one.

BR,
Santiago

SanPen, 2021-04-04:
Today I was trying to make Anaconda/Miniconda work.

Miniconda is possible, Anaconda I was unable to make it work.

The steps are:

- Install Python 3.9 Miniconda3 x64: https://docs.conda.io/en/latest/miniconda.html
    - Should also work for other 3.x versions of python for 64 bits.
    - Uncheck the advanced options during the installation unless you need them.

- Copy the following files from `[miniconda folder]\Library\bin` to `[miniconda folder]\Library\DLLs`
    - libcrypto-1_1-x64.dll
    - libcrypto-1_1-x64.pdb
    - libssl-1_1-x64.dll
    - libssl-1_1-x64.pdb

- Install GridCal: `python.exe -m pip install GridCal`

- Copy the following files from `[miniconda folder]\Library\bin` to `[miniconda folder]\Library\DLLs`
    - sqlite3.dll

- Test: `python.exe -c "from GridCal.ExecuteGridCal import run; run()"`

But as I've said I 100% discourage the use of both Anaconda/Miniconda.

YaeSakura1, 2021-04-07:
Thank you again sincerely！！
I get rid of Anaconda and use regular python 3.9.4 ，and i successfully install GridCal and run it
Thank you for your help！！

YaeSakura1, 2021-04-07:
I follow the tutorials to creates the five-node grid ,but i get this error:
C:\Users\Administrator\AppData\Local\Programs\Python\Python39\python.exe C:/Users/Administrator/GridCal/5node.py
Traceback (most recent call last):
  File "C:\Users\Administrator\GridCal\5node.py", line 32, in <module>
    grid.add_branch(Branch(bus1, bus2, 'line 1-2', r=0.05, x=0.11, b=0.02))
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python39\lib\site-packages\GridCal\Engine\Devices\branch.py", line 304, in __init__
    EditableDevice.__init__(self,
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python39\lib\site-packages\GridCal\Engine\Devices\editable_device.py", line 66, in __init__
    self.idtag = idtag.replace('_', '').replace('-', '')
AttributeError: 'Bus' object has no attribute 'replace'

SanPen, 2021-04-07:
Hi,

The issue comes by the use of Branch instead of Line. 

Could you please tell me where did you see the tutorial so I can update it?

BR,
Santiago

YaeSakura1, 2021-04-07:
I saw the tutorial at https://gridcal.readthedocs.io/en/latest/tutorials/five_node_grid.html

SanPen, 2021-04-07:
This should be the right one:

```
from GridCal.Engine import *

np.set_printoptions(precision=4)
grid = MultiCircuit()

# Add the buses and the generators and loads attached
bus1 = Bus('Bus 1', vnom=20)
# bus1.is_slack = True
grid.add_bus(bus1)

gen1 = Generator('Slack Generator', voltage_module=1.0)
grid.add_generator(bus1, gen1)

bus2 = Bus('Bus 2', vnom=20)
grid.add_bus(bus2)
grid.add_load(bus2, Load('load 2', P=40, Q=20))

bus3 = Bus('Bus 3', vnom=20)
grid.add_bus(bus3)
grid.add_load(bus3, Load('load 3', P=25, Q=15))

bus4 = Bus('Bus 4', vnom=20)
grid.add_bus(bus4)
grid.add_load(bus4, Load('load 4', P=40, Q=20))

bus5 = Bus('Bus 5', vnom=20)
grid.add_bus(bus5)
grid.add_load(bus5, Load('load 5', P=50, Q=20))

# add branches (Lines in this case)
grid.add_line(Line(bus1, bus2, 'line 1-2', r=0.05, x=0.11, b=0.02))
grid.add_line(Line(bus1, bus3, 'line 1-3', r=0.05, x=0.11, b=0.02))
grid.add_line(Line(bus1, bus5, 'line 1-5', r=0.03, x=0.08, b=0.02))
grid.add_line(Line(bus2, bus3, 'line 2-3', r=0.04, x=0.09, b=0.02))
grid.add_line(Line(bus2, bus5, 'line 2-5', r=0.04, x=0.09, b=0.02))
grid.add_line(Line(bus3, bus4, 'line 3-4', r=0.06, x=0.13, b=0.03))
grid.add_line(Line(bus4, bus5, 'line 4-5', r=0.04, x=0.09, b=0.02))

options = PowerFlowOptions(SolverType.NR, verbose=False)
power_flow = PowerFlowDriver(grid, options)
power_flow.run()

print('\n\n', grid.name)
print('\t|V|:', abs(power_flow.results.voltage))
print('\t|Sbranch|:', abs(power_flow.results.Sf))
print('\t|loading|:', abs(power_flow.results.loading) * 100)
print('\terr:', power_flow.results.error)
print('\tConv:', power_flow.results.converged)

grid.plot_graph()
plt.show()
```

YaeSakura1, 2021-04-07:
ok,thank you, but follow the tutorial :Create a grid and run a time series power flow ,it fault again
tutorial at https://gridcal.readthedocs.io/en/latest/tutorials/power_flow_script.html
C:\Users\Administrator\AppData\Local\Programs\Python\Python39\python.exe C:/Users/Administrator/GridCal/5node-.py
Traceback (most recent call last):
  File "C:\Users\Administrator\GridCal\5node-.py", line 8, in <module>
    from GridCal.Engine.Devices.types import BranchType
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python39\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
ModuleNotFoundError: No module named 'GridCal.Engine.Devices.types'

YaeSakura1, 2021-04-07:
Hi,
If I need to use GridCal to implement LCC-MMC three-terminal or multi-terminal HVDC transmission system, is it possible?

SanPen, 2021-04-07:
I believe it is possible. I read somewhere that all converters can be modelled statically using the VSC.

Regarding the tutorial, I'll have a look as well.

SanPen, 2021-04-07:
Also I have just fixed the tutorial

YaeSakura1, 2021-04-08:
Hi,
But if I want to use different control strategies for different converters, the control strategies for LCC and VSC are somewhat different.
For LCC ,it might need to define the firing angle and extinction angle ,is it right?
I read this in Milano's book (Power System Modelling and Scripting)

SanPen, 2021-04-08:
Hi, 

I have that book too. If you point me to the chapter I'll have a look.

Regarding what it is modeled in GridCal, during 2019 and 2020 I read all the papers about HVDC power flow in IEEE and some other journals. The only ones that made sense to me are the following:

- [1] [A generalized frame of reference for the incorporation of, multi-terminal VSC-HVDC systems in power flow solutions](https://www.sciencedirect.com/science/article/abs/pii/S0378779616300566)
- [2] [Flexible General Branch Model Unified Power Flow Algorithm for Future Flexible ACDC Networks](https://ieeexplore.ieee.org/document/8493705)

Why? because these elegantly propose the continuous coupling of AC/DC. Before them the AC and the DC systems were treated as separate grids, simulated independently and then coupled in a separated loop. Way too complicated and failure prone.

GridCal implements and enhanced version of [2] thanks to the help of the authors. 

Now, back to your question, I don't know how to specifically simulate LLC with the VSC model because I cannot find the reference where I read that the LLC static model is a particular case of the VSC static model. It was something along the lines of setting some parameter to zero. 

I hope it helps :/

YaeSakura1, 2021-06-08:
Hi,
sorry for the late reply
Milano's book (Power System Modelling and Scripting),Chapter 17 - Direct-Current Devices ,and Chapter 18 - AC/DC Devices.
Why all converters can be modelled statically using the VSC ?
I read the two papers that you gave me, but these are only VSCs.
I read a paper about LCC-VSC hybrid transmission system:
[1]A sequential power flow algorithm for AC/DC networks incorporating hybrid HVDC
So i want to use GridCal to implement it, then study the electromechanical transient process

SanPen, 2021-06-08:
Hi, 

I read the reference you've mentioned. From what I see there, I still believe that in steady-state the LLC is a simplification of the VSC steady state model. 

If you take the classic reference book [Computer Analysis of Power Systems](https://onlinelibrary.wiley.com/doi/book/10.1002/9781118878309), chapter 4 you'll see a more comprehensive derivation of a sequential power flow. Moreover, I'd say that de decision variables of such LLC system are a subset of the decision variables of the VSC system programmed in GridCal.

Nonetheless, you're 100% welcome to program a new method using GridCal. 

However, consider the following: How are you going to model the LLC converter? Papers like the reference you provided lead me at the beginning of my literature study to think that I should consider the converter as a special type of node that delimits the decoupling frontier between AC and DC. After much study and consideration I found that a far better approach is to consider the converter as a branch connected between an AC and a DC node, which is what I ended up doing.

best regards,
Santiago

YaeSakura1, 2021-06-20:
Hi!

I want to consider the converter as some kind of special node, such as Slack, PQ, PV, just like the bus in AC system. I think the converter ac bus is the boundary between AC system and DC system.

Where did you see that LCC static model is a particular case of the VSC static model or a simplification of the VSC? could you tell me?

Is there any book or paper on modeling of LCC and VSC that I can recommend to read? I can read about it and think about the model.
I also have a question, how should I model the controller? if I want to implement constant DC voltage(Udc), constant DC current(Idc), constant DC power(Pdc), constant firing angle/extinction angle control(α，γ) and the dc voltage droop control, etc

Sincere thanks.

closed 2021-09-09 by SanPen