SanPen/VeraGrid #133: Numerical convergence

andrejcampa, 2021-04-07:
Hi 

Thanks for the software. We have created the radial case grid but have a problem with the numerical convergence. In the time series analysis, for some cases, it converges and in some cases, it does not. For one of the scenarios that do not converge, I have tried to understand what is the main issue. You can get some converge with the numerical method but it converges to the wrong value. It looks like the Linear DC does not converge, but it produces a good starting point. Unfortunately, it does not start or retry the second solver.

I have a few questions and maybe some idea to enhance the software and improve the convergence:
1. Does the restart takes the starting point from the failed method or not? 
2. If the previous is not true, this could be a great enhancement if we could use some kind of slow method to produce some rough estimate of the solution and then use it in one of the fast and NR based methods. So the idea is you could specify 10 iterations of Linear DC and 40 iterations of NR based method in the power flow settings.
3. Why the iterations are always 1 for the NR method?

Br Andrej

SanPen, 2021-04-07:
Hi Andrej!

Thank you for writing.

Not sure how are you reaching to those results. I use GridCal at work with the grid of Spain Portugal and France all together and it works great.

Answering your questions:

1. Yes, on the assumption that it will correct it. The sequence pases through the HELM method and if that method does not converge the grid is wrong 95% sure. You can also check the condition of the grid with the sigma analysis or the voltage collapse functionality.
2. The NR algorithm includes line search which kind of guarantees non divergence. If you're getting divergences with NR, the chances are that the impedances are not ok.
3. The iterations for NR method are the number you specify in the GUI, certainly not 1.

Also, please make sure you're up to date. The current version is 4.0.4.

I can offer you a call to figure it out which maybe is the best way. Please write to santiago.penate.vera@gmail if you want to schedule such a call.

Best regards,
Santiago

andrejcampa, 2021-04-07:
Thanks for a prompt response. Indeed it was a problem with settings of impedance. I will close this issue. There are some glitches when applying settings. We have identified some of them, and we will come back. I am closing this issue since it is not related to numerical methods. Thanks

closed 2021-04-07 by andrejcampa