SanPen/VeraGrid #134: Transformer options

SodinD, 2021-04-07:
Hey, 

I am having some issues with transformers, that I was hoping you could help me resolve.
1.) I cannot delete any of the existing models of transformer in Types Catalogue/Transformers
2.) When I try to change the nominal power of transformer either in the Types catalogue or in Schematic, the values appear to change, however when I run the Powerflow time series, the program still seems to be calculating with the old (previous) values. Also, if I check under Time events ->Transformer -> Rate the values are indeed still set to previous values.

Best regards,
Denis

SanPen, 2021-04-07:
Alright!

I'll have a look

BR,
Santiago

SanPen, 2021-04-07:
Hi, 

I'm on version 4.0.5 and I have not been able to reproduce these issues.
I was able to delete the catalogue objects and I was able to change the snapshot and time series transformer ratings.

Do you have any means that I may reproduce your issue?

BR,
Santiago

SodinD, 2021-04-08:
Hey

I am working with version 4.0.4.

I've created simple network with 2 busses, a load and a transformer. For your convinience I am also attaching the file.
First I have created transformer with values from tutorial (Sn = 0.016, Pcu = 0.45, Pfe = 0.11, Io = 0.68751, Vsc =3.75) and added it to Types catalogue. Later I tried changing its nominal power to 0.2 MVA in Shematic and Types catalogue.
The issue for me is that when I run Power Flow the loading of transformer is 5.9% and when I run Power flow time series the loading of transformer is 70.1% for the same power transferred through transformer (see Figure). To me it looks like the Power flow takes into account new value of Transformer's nominal power 0.2 MVA, whereas the Power flow time series still calculates with the old value 0.016MVA.

Regarding deleting the catalogue item I get following message:
Traceback (most recent call last):
  File "C:\GridCal\python\lib\site-packages\GridCal\Gui\Main\GridCalMain.py", line 4436, in delete_from_catalogue
    used_templates = self.circuit.get_used_templates()
  File "C:\GridCal\python\lib\site-packages\GridCal\Engine\Core\multi_circuit.py", line 1811, in get_used_templates
    if obj.device_type == DeviceType.TowerDevice:
AttributeError: 'NoneType' object has no attribute 'device_type'

Best regards,
Denis

SanPen, 2021-04-08:
Hi, 

The file did not attach :/
You may send it to my email as well.

BR,
Santiago

SanPen, 2021-04-08:
Hi Denis, 

The confusion is the following:
- When you simulate a power flow the schematic colours with the snapshot (makes sense...)
- When you run the time series the schematic colours with the worst loading scenario. For the case you sent me it is ~13% loading:

[image]
I get this is by right clicking on the transformer > plot profiles.

About deleting the template, I rightfully can't do it since it is used in an object. I do not get any error though.

BR,
Santiago

closed 2021-09-09 by SanPen